SpikeInterface/spikeextractors #18: Proposal: RX.addEpoch(name, start_frame, end_frame)

magland, 2018-10-05:
Here's a proposal for a new method in RecordingExtractor (and perhaps this can serve as a template for proposing modifications to spikeinterface)

Proposed actions:
* getEpochNames and getEpochInfo are no longer overloaded by subclass
* add method: addEpoch(epoch_name, start_frame, end_frame)
* RecordingExtractor would keep track of the epochs internally
* Modify MultiRecordingExtractor to use the new addEpoch method in its constructor

Rationale:
It may be useful to add epoch information to recordings that were loaded from a format that does not support epochs. This therefore provides more flexibility. Furthermore it simplies subclasses (because they do not need to keep track of epochs).

Related information:

There are 3 types of methods in the extractors:
* Required to be implemented in subclass (@abstractmethod)
* May optionally be implemented in the subclass
* Should not be implemented in the subclass

colehurwitz31, 2018-10-08:
I agree with this, I can put this in to the main class. So the main class will currently have:

- addEpoch(epoch_name, start_frame, end_frame)
- getEpoch(epoch_name)

I still think we need two more functions, however, to make this complete:

- removeEpoch(epoch_name)
- getEpochs()

The reason for removeEpoch is if you set epochs that you no longer care about later, it should be easy to remove them. The reason for getEpochs is if you set a bunch of epochs, there should be an easy way to return how many you have set and what their frame ranges are.

magland, 2018-10-08:
I agree with removeEpoch. But rather than getEpochs, I think it is better to do:
getEpochNames() -- returns a list of epoch names
getEpoch(name) -- returns the dict containing start_frame and end_frame fields

colehurwitz31, 2018-10-08:
I was thinking that getEpochInformation or getEpochs could allow the user to see which epochs are overlapping or what current epoch ranges they have. If you only have getEpochNames, then you have to get the names and then get each epoch to see how they look which would be cumbersome.

magland, 2018-10-08:
Yes it may be cumbersome, but I have learned to program APIs such that there is one and only one way of doing each thing -- and then there can be convenience functions. But having multiple ways to retrieve the same information can be problematic -- and requires more complicated testing.

Also, maybe getEpochInfo is better than getEpoch for retrieving the dict of a single epoch.

colehurwitz31, 2018-10-08:
Ok, I agree with what you are saying. So the functions we could have are:

- addEpoch(epoch_name, start_frame, end_frame)
- getEpoch(epoch_name) --> returns SubRecordingExtractor
- getEpochNames() --> returns list of names

Could getEpochNames be changed to getEpochInfo which returns a dictionary of names and frames?

magland, 2018-10-08:
I think it should be

getEpochInfo(name)

which returns the dict for just that one epoch.

colehurwitz31, 2018-10-08:
Sounds good. I can try implementing that then.

colehurwitz31, 2018-10-09:
If I construct a SubRecordingExtractor manually and get it an epoch name, should the RecoringExtractor store that epoch name/info?

Also, should we let epochs specify which channels it is tracking?

magland, 2018-10-09:
For second question, i think epochs are related to time, so I would say no.

In general, I don't think subrecordings need to be aware of where they came from, because that info can be kept track of by the program that is using the API.

For example:

X=Y.getEpoch('some_name')

Then if later I wanted to know the start and end frames:

Y.getEpochInfo('some_name')['start_frame']

See, it doesn't need to be stored with X.

colehurwitz31, 2018-10-09:
So would have:

-   addEpoch(epoch_name, start_frame, end_frame)
-   getEpoch(epoch_name) --> returns SubRecordingExtractor
-   getEpochInfo() --> returns dictionary of all epochs which has 'epoch_name': (start_frame, end_frame)

magland, 2018-10-09:
Yes, but I think there should be only one epoch which has a given name. So I'm confused by the phrase "all epochs".

colehurwitz31, 2018-10-09:
Currently, I am storing one dictionary that has all epoch_names as keys and each item is a key and tuple of start and end frames. I can return that whole dictionary so the user can get the names and start and end frames for their epochs.

colehurwitz31, 2018-10-09:
Could you also explain this code:
```
if epoch_name is not None:
            e_info=parent_extractor.getEpochInfo(epoch_name)
            e_start=e_info['start_frame']
            e_end=e_info['end_frame']
            if start_frame is None:
                start_frame=e_start
            else:
                start_frame=e_start+start_frame
            if end_frame is None:
                end_frame=e_end
            else:
                end_frame=e_start+end_frame
```

magland, 2018-10-09:
This is in the constructor of the SubRecordingExtractor right?

If the user has specified an epoch name, then we get the info in the parent extractor for that epoch name... and use that start time and end time. But if the user also specified a start_frame and/or end_frame (in addition to epoch_name), then we assume they are requesting a start_frame and end_frame relative to the epoch.

magland, 2018-10-09:
> Currently, I am storing one dictionary that has all epoch_names as keys and each item is a key and tuple of start and end frames. I can return that whole dictionary so the user can get the names and start and end frames for their epochs.

I don't think it is a good idea to return that dictionary. That's the internal way you store it, but should not be exposed in the API.

magland, 2018-10-09:
oh i see... I misunderstood the above.... it should be
getEpochInfo(epoch_name)

which returns the epoch info just for that epoch.

magland, 2018-10-09:
And it should return a dict:
{
  'start_frame':[the-start-frame],
  'end_frame':[the-end-frame]
}

colehurwitz31, 2018-10-09:
I was thinking it could return a copy of the internal dictionary since that is the easiest way to return all names and frame information.  I would rather return all the info otherwise we would need a getEpochNames so that they could figure out which epochs they have set.

Also, I was wondering if we should remove all references to epochs in the SubRecordingExtractor. I don't think that a user should initialize a SubRecordingExtractor and have information from the parent extractor added. I think that the only way to get epochs should be through the getEpoch functionality.

colehurwitz31, 2018-10-09:
Currently, the SubRecordingExtractor never uses the epoch_name and I think initializing with an epoch_name that only checks for the epoch name in the parent extractor is confusing since I may just want to name my SubRecordingExtractor and when I set epoch_name it doesn't do anything and I can't return the name

magland, 2018-10-09:
I agree with your second point.

Regarding the first, it is important not to expose the internal structure of how all epochs are stored. There are various reasons for that and it is better to have a getEpochNames() function. I can give more details if you want. It comes from clean api design principles.

colehurwitz31, 2018-10-09:
Current implemenation:
```python
def __init__(self):
        self._epochs = {}
def addEpoch(self, epoch_name, start_frame, end_frame):
        '''This function adds an epoch to your recording extractor that tracks
        a certain time period in your recording. It is stored in an internal
        dictionary of start and end frame tuples.

        Parameters
        ----------
        epoch_name: str
            The name of the epoch to be added
        start_frame: int
            The start frame of the epoch to be added (inclusive)
        end_frame: int
            The end frame of the epoch to be added (exclusive)

        '''
        #Default implementation only allows for frame info. Can override to put more info
        self._epochs[epoch_name] = {'start_frame': int(start_frame), 'end_frame': int(end_frame)}

    def removeEpoch(self, epoch_name):
        '''This function removes an epoch from your recording extractor.

        Parameters
        ----------
        epoch_name: str
            The name of the epoch to be removed
        '''
        if(epoch_name in list(self._epochs.keys())):
            del self._epochs[epoch_name]
        else:
            raise ValueError("This epoch has not been added")

    def getEpochNames(self):
        '''This function returns a list of all the epoch names in your recording

        Returns
        ----------
        epoch_names: list
            List of epoch names in the recording extractor
        '''
        epoch_names = list(self._epochs.keys())
        return epoch_names

    def getEpochInfo(self, epoch_name):
        '''This function returns the start frame and end frame of the epoch
        in a dict.

        Parameters
        ----------
        epoch_name: str
            The name of the epoch to be returned

        Returns
        ----------
        epoch_info: dict
            A dict containing the start frame and end frame of the epoch
        '''
        #Default (Can add more information into each epoch in subclass)
        if(epoch_name in list(self._epochs.keys())):
            epoch_info = self._epochs[epoch_name]
            return epoch_info
        else:
            raise ValueError("This epoch has not been added")

    def getEpoch(self, epoch_name):
        '''This function returns a SubRecordingExtractor which is a view to the
        given epoch

        Parameters
        ----------
        epoch_name: str
            The name of the epoch to be returned

        Returns
        ----------
        epoch_extractor: SubRecordingExtractor
            A SubRecordingExtractor which is a view to the given epoch
        '''
        if(epoch_name in list(self._epochs.keys())):
            from .SubRecordingExtractor import SubRecordingExtractor
            epoch_info = self._epochs[epoch_name]
            start_frame = epoch_info['start_frame']
            end_frame = epoch_info['end_frame']
            return SubRecordingExtractor(parent_extractor=self, start_frame=start_frame,
                                         end_frame=end_frame)
        else:
            raise ValueError("This epoch has not been added")
```

I think this is about right
```

colehurwitz31, 2018-10-09:
The formatting gets a bit screwed up in the markdown (the paragraphs are off)

colehurwitz31, 2018-10-09:
Done! I also made separate checks to make sure epoch names are strings throughout with value errors if not.

The getEpochInfo usage was clever!

colehurwitz31, 2018-10-09:
I actually just realized it should probably return epochs in chronological order. So sort by earliest start_frame

magland, 2018-10-09:
Yes, that's better.

closed 2018-10-10 by magland