SpikeInterface/spikeextractors #28: Unit Tests insufficient for getSnippets and getTraces

colehurwitz31, 2018-10-09:
Currently, the unit tests only check if the getSnippets and getTraces functions are equal for two recording extractors. However, the functionality could be the same for both, but still be wrong. I think I just removed a bug from getSnippets which was not being caught since the core implementation was wrong for buffering near the ends of files.

colehurwitz31, 2018-10-09:
Or at least the getSnippets test is insufficient as the previous code should not have been working for end of file issues.

I just changed:

```
if snippet_range[1] >= num_frames:
                    snippet_buffer[1] -= snippet_range[1] + num_frames
                    snippet_range[1] -= snippet_range[1] + num_frames
```

to:

```
if snippet_range[1] >= num_frames:
                    snippet_buffer[1] -= snippet_range[1] - num_frames
                    snippet_range[1] -= snippet_range[1] - num_frames
```

magland, 2018-10-09:
Nice catch. For this reason, I have a test called test_example which makes sure that the example dataset itself is working, but I think that it did not include getSnippets or getRaw test. We'll need to make sure we cover all the bases in the test_example.

colehurwitz31, 2018-10-09:
Yes, I agree. Would you like to work on the test_example or are you currently working on something else? I am working on the getUnitProperty issue currently. Trying to get all SpikeInterface issues solved today so that I can begin writing about SpikeInterface

colehurwitz31, 2018-10-09:
(Not including subclasses since those will take a while and are separate from the core of SpikeInterface.

magland, 2018-10-09:
Yes, I will work on the unit tests. First I am trying to finalize MountainSort alg using spikeinterface.

closed 2018-10-24 by colehurwitz31