SpikeInterface/spikeextractors #29: SubSortingExtractor and MultiSortingExtractor -- Dealing with num_frames and other info

colehurwitz31, 2018-10-09:
I can't remember if we came up for a solution for this, but currently the SubSortingExtractor and MultiSortingExtractor are non-functional and having RecordingExtractors being passed in for information for the recording.

magland, 2018-10-09:
non-functional? what do you mean by that?

colehurwitz31, 2018-10-09:
Whoops, I actually fixed the issue of the SubSortingExtractor not working earlier (its parent extractor was calling both Recording and Sorting extractor functions)

I think they are both functional, but untested. Also, did we have a solution for the MultiSortingExtractor not having RecordingExtractors added to it since the SubRecordingExtractor does not have a RecoringExtractor?

magland, 2018-10-09:
There is a `test_multi_sub_extractor`, so I think it is tested at least to some extent.

I don't understand your question: `did we have a solution for the MultiSortingExtractor not having RecordingExtractors added to it since the SubRecordingExtractor does not have a RecoringExtractor?` could you please explain a bit more?

magland, 2018-10-09:
First of all, there is something troubling me about this. I don't think SubSortingExtractor should have total_recording_frames.

colehurwitz31, 2018-10-09:
I thought we agreed that we didn't want RecordingExtractors being passed into SortingExtractors. We wanted a way to pass in the frames that a SortingExtractor encapsulated for different analysis and plotting

colehurwitz31, 2018-10-09:
I thought we agreed to that so I implemented SubSortingExtractor with frames passed in rather than a RecordingExtractor. I made a pull request with it, but maybe I forgot to talk about that change in my summary

magland, 2018-10-09:
Okay I remember. But there is still something bothering me about it in the context of SubSortingExtractor. Could you give me call on hangouts when you have a chance and we can discuss?

closed 2018-10-12 by colehurwitz31