SpikeInterface/spikeextractors #30: getUnitSpikeTrain in NumpySortingExtractor and MDASortingExtractor return ints and floats respectively.

colehurwitz31, 2018-10-09:
It looks like MDASortingExtractor is returning a numpy array of floats (although frames should be ints)

colehurwitz31, 2018-10-09:
This can be checked for with static typing which I aim to do this week

magland, 2018-10-09:
Actually, I think it should return floats, since we can have events at times that don't line up with the time grid. This can happen, for example, if we upsample to get more precise detection.

colehurwitz31, 2018-10-09:
I thought getunitSpikeTrains always returned frames which would not have any decimal values?

magland, 2018-10-09:
It is units of frames, but you could have an event that occurs between two timepoints (timeframes), if the detection precision is greater than the sampling rate.

colehurwitz31, 2018-10-09:
I see. So that is why getSnippets has integer wrappers around the center frames... that makes sense I guess. I find the concept of frames having decimal values to be confusing, but I understand the use case. So then NumpySortingExtractor should probably be updated to floats.

magland, 2018-10-09:
Yeah, we do use this sometimes.

colehurwitz31, 2018-10-09:
Ok, so we should make sure all extractors do what MDASortingExtractor does

closed 2018-10-24 by colehurwitz31