SpikeInterface/spikeextractors #33: Proposal: setChannelProperty/getChannelProperty rather than getChannelInfo

magland, 2018-10-10:
In RecordingExtractor, some methods are overridable (expected to be overwritten in a subclass), and some are not. There are at least two reasons to make a method overwritable. But I think in the case of getChannelInfo, neither one of them apply. Therefore, to conform with the getUnitProperty/setUnitProperty method of SortingExtractor, I propose we replace getChannelInfo by

setChannelProperty(property_name, value)
getChannelProperty(property_name)

One special property will be 'location', which returns a 2-vector or 3-vector with the electrode coordinates.
Another property will be 'group', an integer specifying the group of the electrode.

magland, 2018-10-10:
Note that setChannelProperty will often be called in the constructor of the subclass, just like addEpoch is called in the MultiRecordingExtractor

colehurwitz31, 2018-10-10:
Same implementation and naming scheme as the unit property then?

alejoe91, 2018-10-10:
I agree too

closed 2018-10-12 by colehurwitz31