SpikeInterface/spikeextractors #34: Proposal: getUnitSpikeFeatures(name, unit_id, start_frame, end_frame)

magland, 2018-10-10:
The getUnitProperty() method is great. However, for large recordings, and for compatibility with getUnitSpikeTrain, we will need an alternative that parallels the getUnitSpikeTrain, in that it can efficiently retrieve data for a range, start_frame, end_frame. 

I propose to make getUnitSpikeFeatures, which is overridable in subclasses and has the same interface as getUnitSpikeTrain() (plus the feature name)

For example getUnitSpikeFeatures('location', unit_id=1, start_frame=0, end_frame=10000)
could return a 2x50 array of the 2-d location of the event (as determined by the sorting algorithm or some other method). Note that the corresponding getUnitSpikeTrain would need to return the a vector of agreeing size (50). 

This could also hold any other per-event features (outlier score, pca, etc)

colehurwitz31, 2018-10-10:
Possible confusion with this method is that I may have features that are dependent on the channel. For example, I may detect 4 waveforms for one spike and then extract PCA features from each channel and cluster. 

I am not sure how we can work that in

magland, 2018-10-10:
True... we can think about it.

closed 2018-10-24 by colehurwitz31