SpikeInterface/spikeextractors #43: Argument and file name standardization for the tools.py file

colehurwitz31, 2018-10-15:
So the argument name for a bunch of functions in the recording extractor in the tools.py file is 'recording'. I think this should be changed to 'recording_extractor' since that is consistent with the rest of the repo.
I also think it is confusing to have 'recording' be an extractor rather than a file. 

Also, should we call this a tools file? It only seems to deal with probe information so I would rather have it be called probe tools or something since we may have other tools in the future which we don't want in the same file.

colehurwitz31, 2018-10-15:
I will change these for now and create a pull request

alejoe91, 2018-10-15:
@colehurwitz31 can we close this?

closed 2018-10-15 by alejoe91