SpikeInterface/spikeextractors #55: deal with channel map

alejoe91, 2018-10-24:
the probe file can also provide a channel map, i.e. how the channels should be reordered (for example in a linear probe) to match the linear geometry. 
I think that in the recording extractor we could have a self._channel_map which is None by default, and if a channel map is present, then `getTraces` returns:

```
if self._channel_map is None:
    return getTraces()
else:
    return getTraces()[self._channel_map]
```

the channel map can be loaded from the probe file using the loadProbeFile function

alejoe91, 2018-10-24:
alternatively, we could have a `remapChannels()` function, that reorders the channels..but that would not allow for memory mapping..
My soution would be having a  `_getTraces()` as abstract method, and then implementing in RecordingExtractor:
```
def getTraces(start_frame=None, end_frame=None, channel_ids=None):
    if self._channel_map is None:
    return self._getTraces(start_frame, end_frame, channel_ids)
else:
    return self._getTraces(start_frame, end_frame, channel_ids)[self._channel_map]
```

colehurwitz31, 2018-10-24:
I am not sure this is necessary and would be problematic (for example, if I specify channel ids in getTraces, would the channel map change?)

alejoe91, 2018-10-24:
I believe it is necessary. I'm actually finding troubles now because we are using a probe with a screwed up channel map, and it would be great to adjust it before spike sorting. Channel_ids should then be just linear in my opinion:
assume you have 4 channels, `self._channel_map` could be `[3, 1, 0, 2]`
then `getTraces(channel_id=[0, 1])` would return the original channel 3, 1

colehurwitz31, 2018-10-24:
channel_map could be a good variable to have then. It sounds useful. @magland ?

magland, 2018-10-24:
I think self._channel_map breaks the discipline we have achieved thus far. I think this should rather be an input to the load probe function. The important thing is that the channel_id corresponds with the channels in the recording. Thus the channel ids should be modified from the original probe if needed. If you want to retain information about the original channel id, then that can be a channel property. (That is my opinion).

So what I am saying is that channel_id maybe should be channel_index throughout... and it always corresponds to the order of data in the underlying recording.

colehurwitz31, 2018-10-25:
Agreed. That makes good senses. The probe functions can make the channel ids match our convention.

alejoe91, 2018-10-25:
@magland Why is it breaking the discipline? I think it could actually be very useful. 

*I think this should rather be an input to the load probe function. The important thing is that the channel_id corresponds with the channels in the recording. Thus the channel ids should be modified from the original probe if needed. If you want to retain information about the original channel id, then that can be a channel property.*

I agree this information should be input to the load probe function, but if it's not saved in the recording extractor, then it's lost. Maybe I did not understand what you mean with this.

Let me make an example: if the channel order is mixed, (and this is unfortunately the case form many commercial probes, including the ones that we ar eusing now) it would be hard to visualize neighboring channels using the TimeSeriesWidget. Instead, giving the possibility to add a channel map would make this easier. The same holds for giving the geometry: I think it would be much easier and less prone to in the probe file to sequentially describe the geometry rather than write down the mixed geometry.

It could be actually ok to use a channel_id property, loaded from the probe file, but I believe that in that case all the spike sorters should reorder the recordings according tho this property (if present) and the same holds for wigets etc..

colehurwitz31, 2018-10-25:
I think Jeremy is saying that we always return channels in the way we defined. However, if a probe is supposed to be visualised in a different order, you can make a dictionary or something that maps the returned channel ids to the correct channels for visualisation. 

So you can return 0 - 384 channels for some arbitrary probe, but then 0 —> 10, 1 —> 20, 2 —> 30, etc.

alejoe91, 2018-10-25:
the problem is that channels are not defined. we only have a `getNumChannels` and assume that they are from 0 to `getNumChannels` - 1. 

I think that the `getChannelIds` could be a good solution, as it does not require the user to manually map the channels all the times and it's at the user's discretion tu use the channel map (`getChannelIds`) or not. What do you guys think?

magland, 2018-10-25:
I was suggesting that we use the getChannelProperty for this. For example

getChannelProperty(channel_id=1, property_name='order')

But maybe getChannelIds() could work. Certainly easier for the widget development side. But to be more clear on what it is, I would maybe suggest:

getChannelOrdering()

which by default returns [0,1,2,3,...,M-1]

but can return a different ordering. How does that sound?

alejoe91, 2018-10-25:
Yep could work. 
Regarding the naming, I personally like `getChannelIds()` for 2 reasons:
- as counterpart to `getUnitIds()`. 
- We are going to be using it a lot so a few less characters are welcome :) 

Also, thign about how many times in spiketoolkit we need `range(recording.getNumChannels())`: that would become: `recording.getChannelIds()`

colehurwitz31, 2018-10-25:
I keep flip-flopping, but Alessio's idea seems good.

If we implement getChannelIds() and have a default map equal to (0 - num_channels), that map can be set and get later.

alejoe91, 2018-10-25:
@magland if you agree, I would suggest to have a `self._channel_ids` that is None by default (in that case `getChannelIds()` would return `list(range(recording.getNumChannels()))` and can be overwritten by `loadProbeInformation`

colehurwitz31, 2018-10-25:
Might as well have self._channel_ids be range*0,  num_channels) by default.

alejoe91, 2018-10-25:
no because you don't know the number of channels by default and self._channel_ids must be instantiated. So I would say that the developer of an extractor  does not need to set it to range(0, num_channels), and getChannelIds can check if it's not None

colehurwitz31, 2018-10-25:
getNumChannels is an abstract method so we would know the number of channels beforehand

alejoe91, 2018-10-25:
Can you `self._channel_ids = list(range(self.getNumChannel()))` in the `RecordingExtractor` init?

colehurwitz31, 2018-10-25:
So getNumChannels can be run in init, although there is an issue with the numpy extractors because they set all the data values and channel information after initialization... all the other extractors should be fine though.

colehurwitz31, 2018-10-25:
Maybe we can have channel_ids be None until it is either set (setChannelids) or if getChannelIds is called and it is None then we can call getNumChannels and set channel ids.

magland, 2018-10-25:
didn't follow the whole discussion, but I think I'm now on board with getChannelIds()
by default it should give [0,...,M-1]

alejoe91, 2018-10-25:
should w ehave a `setChannelIds()` too to set the channel map?

magland, 2018-10-26:
Yes, I think that setChannelIds() would be the mechanism for setting this. And the implementation should verify that the list sent to it is of length getNumChannels() and that it includes each channel id exactly once. Right?

alejoe91, 2018-10-26:
@magland @colehurwitz31  Yep getChannelIds and setChannelIds. Not sure if we should verify the permutation thing. Some devices also load adc signals and other external stuff. Myabe we should verify that channel:ids is a subset of range(getNumChannels()).

magland, 2018-10-26:
Something about getChannelIds and setChannelIds still makes me nervous, and partly because I am having a hard time wrapping my head around it.

One thought... could this all be handled using SubRecordingExtractor? Just put in channel_ids=[1,0,3,2,5,4], or example, in the constructor of SubRecordingExtractor, and we have achieved a new ordering. Also, this way we can exclude channels. This might be cleaner.

colehurwitz31, 2018-10-26:
SubRecordingExtractor can definitely work for changing the channel order. 

I think we need to see how widely people will want to be using this function. If for every probe, people want to return the getTraces in different orders, then we probably want to handle that in RecordingExtractor rather than making people make a RecordingExtractor and then a SubRecordingExtractor.

I am not sure that getChannelIds and setChannelIDs are the right idea in the probe. It is more like getChannelOrder and setChannelOrder to me.

magland, 2018-10-26:
I have a better idea of why I feel uneasy about this... Let's consider a case where we have 8 channels.

R.getNumChannels() is implemented by the subclass. Therefore, it cannot be changed by the parent class. It will always be 8. So, if you do R.setChannelIds([0,2,6,4]), you will have a mismatch between len(R.getChannelIds()) and R.getNumChannels().

But I can see that selecting a subset of channels and reordering of channels is a common case. Therefore, I propose the following:

We remove getNumChannels(), and have getChannelIds() behave just like getUnitIds() -- that is, an abstract method implemented by the subclass. But we cannot have setChannelIds() because getChannelIds() is implemented in the subclass. Therefore, we must use SubRecordingExtractor to get a recording with a subset or reordering of the channels. This is just like SubSortingExtractor used to get a subset (or reordering) of the units.

Even though it requires changing all our code and examples, I'm actually feeling pretty confident that this is what we should do.

magland, 2018-10-26:
(continued) if we do decide to make the above change, then we can

magland, 2018-10-26:
(oops, accidentally closed the issue -- reopening)

alejoe91, 2018-10-26:
I would leave getNumChannels() too, but generally I'm ok with the idea. In that case loadProbeInfo would return a recording extractor. Ok?

colehurwitz31, 2018-10-26:
I don't think there is a need for getNumChannels if we have getChannelIds since the length of that list should be the number of channels. I can implement this and push if we agree.

magland, 2018-10-26:
Sounds good. This will be nice because getChannelIds() will match with getUnitIds(). After you implement, I can help update all the code to conform.

alejoe91, 2018-10-26:
@colehurwitz31 I know there is no need but I think `getNumChannel()` is very useful :) on the same page I would recommend adding a `getNumUnits()` in the Sorting Extractor

magland, 2018-10-26:
@alejoe91. I agree that this would be useful. But here's the danger... somebody might do:

```
for ch in range(R.getNumChannels()):
   do something with ch
```

This would work for most cases, but then would unexpectedly fail when the channel order was non-trivial. So we should think about this carefully before making that decision.

colehurwitz31, 2018-10-26:
Another issue is that getTraces has the channel_ids argument. As long as we have some internal channel_ids variable, should the channel_ids argument order not matter?

colehurwitz31, 2018-10-26:
Could we just the channel_ids argument figure out the order that we return traces?

```
channel_ids = [1,2,0,5,4]
getTraces(channel_ids=channel_ids)

[trace1, trace2, trace0, trace5, trace4]
```

alejoe91, 2018-10-26:
Yeah I think that the behavior then should be the same as units in sorting extractor, in which when we use `getUnitSpikeTrain()` we give the unit id.

alejoe91, 2018-10-27:
Ok, so @colehurwitz31 you do the channelIds? I'll update the loadProbeInfo

alejoe91, 2018-10-27:
Actually @colehurwitz31 and @magland I made an initiall attempt to solve the problem and will make PR.
I kept the getNumChannels for the moment.

alejoe91, 2018-10-27:
https://github.com/colehurwitz31/spikeinterface/pull/57

closed 2018-10-30 by colehurwitz31