SpikeInterface/spikeextractors #74: t_start concept should be introduced in RecordingExtractor

samuelgarcia, 2019-02-08:
In RecordingExtractor.frameToTime we have:
```time = frame / self.getSamplingFrequency()```
in neo we also introduced **t_start** with is the time of the first sample and so we have
```time = frame / sampling_rate + t_start```
even if t_start is moslty 0 and so the same case as here in some situtation it is very convienient for time serie manipulation to have it.

colehurwitz31, 2019-02-08:
Hey Sam,

I will look into this. As it is now, I think the start_frame/start_time of a recording is considered to be 0, so there isn't necessarily a need for a t_start object. Since everything is done with frames in our framework, RecordingExtractor.frameToTime just is a nice convenience function.

colehurwitz, 2019-09-10:
still something we should do, @samuelgarcia ?

samuelgarcia, 2019-09-10:
We can close if you want. I will be back on this after neo integartion in 2023.