SpikeInterface/spikeextractors #75: Semantics about RecordingExtractor and MultiRecordingExtractor

samuelgarcia, 2019-02-08:
Semantically, what you call a "recording" is ONE continuous recording of a multi channel signals.
Symantically, in neo, we do not use "recording" because it can include many different concept.

For instance, in some case (Blacrock, Neuralynx or Axon systems ) a recording can contain some pauses. In that case, the multi channel signal recording is not continuous but contain gaps.
this is why we introduced the **Segment** semantic. If you have 2 pause, you will have 3 Segments.

Here, from what I guess from the code, **RecordingExtractor** deal only with one "continuous recording" and **MultiRecordingExtractor** deal several "continuous recording".
My very first impression : you should **not** have 2 differents classes. **RecordingExtractor** should always deal with several "Segment" even if the main case is a list one "recording".

Furthermore, the multi recording should be totally explicit and not "automagic".
So the ```MultiRecordingExtractor.getTraces``` should have epoch_index or epoch anme kargs.

In the actual case, **MultiRecordingExtractorgetTraces** do a in **memory** concatenate. This is totally insane from a performance view. This is also a mistake from the filetring view because signal borders lead to filter artifact and by the concatenating you introduce fake borders.

colehurwitz31, 2019-02-08:
Hi Sam,

RecordingExtractors were designed to extract information from a _single file_ containing raw extracellular data. This covers most of the basic cases for raw data formats used in experimental labs. For a raw data format that has multiple pauses (but is contained in one file), a RecordingExtractor would be designed to read and deal with these pauses.

MultiRecordingExtractors allow for extracting from _multiple files_ containing one raw extracellular data. They actually inherit from the RecordingExtractor base class. Since we wanted all RecordingExtractors to have the same functionality, we made getTraces (and a few other select functions) abstract methods, which means that they must be called in the same way in each subclass. Therefore, MultiRecordingExtractor.getTraces is the same as the RecordingExtractor.getTraces. This was a conscious design decision. 

For the concatenation concerns, I am not sure what you mean by introducing fake borders, we have not run into that issue. The MultiRecordingExtractor expects each file to part of the same recording and continuous so concatenation makes sense at the end. We have also not run into the memory issues that you mentioned, but I will look into that just in case.

alejoe91, 2019-02-08:
@samuelgarcia 
A RecordingExtractor is indeed a semantically similar to a segment, i.e. a single continuous recording. I agree we should add the possibility to combine multiple recordingextractors in a MultiRecordingExtractor with possibility of adding pauses. (In this case the MultiExtractors could be seen as Blocks)
An extra `start_frames` argument would do the trick (writing an issue about that).

Regarding the memory issue, could you elaborate on that? The concatenation happens only if the data are stored on separate files (separate extractors), so I don't know how this could be improved. In case data are retreived from the same recording extractor then only the selected extractor .getTraces is actually called. 
From the filtering point of view, if the recordings are consecutive then it is the correct implementation, if there are pauses (after we added the `start_frames` option), then we should implement filtering on consecutive frames only.

closed 2019-02-22 by colehurwitz31