SpikeInterface/spikeextractors #76: Semantics alignedment with neo

samuelgarcia, 2019-02-08:
This is not really an issue but a general comments.

First, I have to say that I am very antousistast with this project.
This is really great this joint effort.
I plan to help for real.

The first impression (which is not a big issue) is the missalignement with the python-neo semantic.
Note that I am member of neo team (you are all welcome).

From what I anderstand, but maybe I missing some concept:
  * an **Extractor** would be in neo an **IO** with reading capability.
  * an **Recording** would be in neo a AnalogSignal.
  * a **RecordingExatractor** would be in neo a **IO** that read one AnalogSignal.
  * an **MultiRecordingExatractor** would be in neo an IO that read several Segment
     with AnalogSignal in each.
  * an **SortingExtractor** would be a **IO** that read one Segment with several SpikeTrain.
  * an **MultiSortingExtractor** would be a **IO** that read several Segment with several SpikeTrain.

This new semantic for same purpose is not a big deal at the end because I can adapt easily.
But neo semantic (and concept) have been disscussed (and improved) from 10 years now with a quite big rolling team over that period.
Since I and many others am involved in the spike sorting problem, I spend some effort to have in neo all the semantic and concept to represent the spike sorting problem.

So in conclusion, as this project is quite young, I would strongly advice at least to llok at the neo samantic and maybe move toward it. Of course I can help.

I also put Andrew Davison @apdavison @mdenker @JuliaSprenger, in this discussion.

colehurwitz31, 2019-02-08:
Thanks for the comment and ideas. 

We think the critical difference between NEO and SpikeInterface is that they attempt to solve different problems, which affects the implementation. NEO works with all electrophysiological files, loading them into memory and then creating an object-oriented representations of the data, making it more accessible for analysis. SpikeInterface only works with raw and spike-sorted extracellular recordings and provides users direct access to the underlying datasets with standardized extraction functions. This allows for easy integration into and construction of reproducible spike sorting workflows.

As a result, SpikeInterface has a simple inheritance structure (extractors are our only objects) which means that the user only interacts with these objects to retrieve data from files. Data is retrieved and passed as numpy arrays or basic python data structures, allowing for direct manipulation of the data in python. We very deliberately chose this design as it makes designing a pipeline very easy - for instance, spike sorting just requires two lines of code. Our interface design also means that extractors cannot be re-interpreted as neo IO objects, since extractors are used to retrieve and output data without any object-oriented representations.

In addition, SpikeInterface makes the type/format of the data source irrelevant as long as an extractor is available (the path to the data can, for instance, be an URL). Spike sorting requires access to raw and processed data, so two core classes, RecordingExtractors and SortingExtractors, for raw and spike-sorted files respectively, are sufficient for the purpose of workflow integration.

Here we see possible links between the projects, SpikeInterface could operate on top of NEO to provide access to a range of data formats for spike sorting, curation and validation workflows. We should discuss possible bridges between SpikeInterface and NEO in more detail offline, as NEO is a great project that nicely solves some fundamental data portability issues.

@magland @alejoe91 @mhhennig

alejoe91, 2019-02-08:
@samuelgarcia I think that we would like to keep SpikeExtractors as light-weight and easy to learn. In the beginning we discussed about using neo, but hen we opted to keep it as simple as possible, so that also experimental people with little coding bakcground can quickly access the data, run spike sorting, and check and validate the output (see [tutorials](https://github.com/SpikeInterface/spiketutorials)).

On the other hand, we all agree that neo is a powerful tools for IO and organizing the data (we use it a lot in our lab!), but maybe SpikeInterface could come on top of that?

Would you be interested in implementing a NeoExtractror? 

```
import neo
import spikeextractors as se

io = neo.SomeIO('some-file')
recording_extractor, sorting_extractor = se.NeoExtractor(io)
# if the neo object only has analog signals, it returns sorting_extractor=None, and viceversa if it only has 
# spiketrains. If both are available it could load both
```

Regarding the `block`-`segment` issue (I will reply there too), we will add a `start_frames` argument to the multi sorting extractor so that multiple pauses can be accounted for. If the neo object has multiple segments, the returned `recording_extractor` will effectively be a MultiRecordingExtractor, but it is derived from the base RecordingExtractor class so nothing would change downstream (for sorting etc).

samuelgarcia, 2019-02-11:
Hi all.
Thanks for anwsers.
I can try  to make this NeoExtractor. It should be easy in the standard case.