SpikeInterface/spikeextractors #81: dim for trace should be inverted.

samuelgarcia, 2019-02-11:
In actual implementation, when you do a **getTraces()** you have 2D array with (n_chan, n_sample).

This is the intuituive choice BUT for big files this lead to bad performences.

The reverse (n_sample, n_chan) in conetxt of long recording could lead to better performence.

A natural device give sample stream in the (n_sample, n_chan) order.
So many system using raw binary format go for (n_sample, n_chan) on thedisk . So a simple memmap can read the whole file.

Here with (n_chan, n_sample) a full memory copy  and translate must be done after the read.

Further, if you acces the signals by chunk in time, you need to acces bytes that are very distant inside the binary file.

I would highly, suggest to invert this and go for (n_chan, n_sample).

colehurwitz31, 2019-02-12:
Hi Sam,

Inverting the dimensions does seem like a good idea if the performance would be affected otherwise.

Seeing as getTraces is a core function or our framework, I’m gonna give it some thought with the other people on the team about how and when to implement this.

magland, 2019-02-12:
Hi Sam and Cole,
I am open to this possibility, but I want to point out that this could be an internal change (implementation) in how the data are stored internally for better performance -- without changing the API of recording extractors. Some formats will do it one way, and some will do it another -- the interface should be agnostic to that.
Jeremy

colehurwitz31, 2019-02-12:
Hey Jeremy,

I am a bit confused about what you are suggesting. Are you saying that we can leave getTraces untouched, but change how data is stored when we extract it? 

Cole

magland, 2019-02-12:
Yes. The point of an interface is that it abstracts away how data is handled internally. It seems against the philosophy of such a layer to modify it based on such implementation/efficiency optimization. IMO, it does not need to be changed in order to improve how data are stored.

alejoe91, 2019-02-12:
@magland so how do we do it?

magland, 2019-02-12:
There are various recording extractors. They are considered on a case-by-case basis. Which one should we focus on first -- for this issue?

alejoe91, 2019-02-12:
I see, basically changing the specific extractors representation and returning the transpose should work

magland, 2019-02-12:
Exactly. And numpy even has a way to specify the internal ordering of arrays. So you could either return a Fortran- or C-style ordering in the returned traces. For example, see: https://docs.scipy.org/doc/numpy-1.15.0/reference/generated/numpy.asfortranarray.html

magland, 2019-02-12:
So IMO, it's up to the individual extractors on how to do it efficiently. The interface is supposed to be able to  abstractly handle all kinds of formats and storage orders.

colehurwitz31, 2019-02-12:
Sounds good. We can just focus on improving specific extractors then.

closed 2019-02-12 by colehurwitz31

samuelgarcia, 2019-02-12:
Of course with with numpy we can do a ```transpose()``` which internally change the array [strides](http://www.jessicayung.com/numpy-arrays-memory-and-strides/).

The thing is that numpy by default is C order and if by default the getTraces() render (n_chan, n_sample) then by default internally a traces will have continuous samples in memory for each channel.

So by default, unless an internal trick on strides (with transpose) is done every time, the performance will be affected when accessing data by chunk along the time axis (1).

I anderstand, that it is a pain to change this because it break many scripts but the project still is at its infancy...

But I don't want to be the guy whi insist.

It was long discussion inside neo team and my colleagues still think that the time should in dim1 (as here) but I really prefer dim0. Long debate...