SpikeInterface/spikeextractors #82: What is the distance units for PRB in saveProbeFile ?

samuelgarcia, 2019-02-13:
In tridesclous and circus the PRB file at 'geometry'  units are supposed to micrometers. (this is not the case for klusta and kilosort I think)

The real dimension for probe is taken in account by theses two tools at least

Here with "toy_example1" the saveProbeFile give
```
           'geometry':  {
               0: [1.0, 0.0],
               1: [2.0, 0.0],
               2: [3.0, 0.0],
               3: [4.0, 0.0],
           }
```
this is non realistic and can lead to bad results.

Is there in this framework some convetion about the real dimension of the probe ?

colehurwitz31, 2019-02-13:
Hi Sam,

I did not create that example, but I think that toy_example1 was designed as a quick/non-realistic way to show the features that spikeextractors and spikewidgets can do. In that regard, I think the geometry of this fake probe (created in synthesize_random_waveforms.py)  was just made for showing-off widgets rather than for being realistic.

To my knowledge, toy_example1 hasn't been used for sorting or any real analysis. We can adjust our toy example though to be more realistic.

I am not sure what you mean in your final question, but generally, probe files should take into account the real dimension of the probe.

alejoe91, 2019-02-15:
@samuelgarcia I can confirm that the units are in um. In klusta the units does not make a real difference as long as we are consistent with the adjacency radius, because the adiacency graph is computed from it. Also in kilosort I think that the units are in um.

samuelgarcia, 2019-02-15:
Ok thank you.

closed 2019-02-20 by colehurwitz31