SpikeInterface/spikeextractors #87: writing binary files -- need to explicitly state fortran or C order

magland, 2019-02-22:
Internally, numpy can represent n-d arrays either in either Fortran (column major) or C (row major) format [see numpy docs](https://docs.scipy.org/doc/numpy-1.14.5/reference/generated/numpy.array.html). When an array is written out to file -- either using np.save() or file.write() -- you get different results depending on the internal representation. This is dangerous because sometimes an array is passed to a function and it is not known what the internal order is.

Therefore, I suggest to explicitly specify the order whenever writing to binary file.

For example, in the spyking circus wrapper, I see this line:

```
np.save(str(output_folder / file_name), recording.getTraces().astype('float32'))
```
We should specify order='F' as follows:

```
np.save(str(output_folder / file_name), recording.getTraces().astype('float32', order='F'))
```

alejoe91, 2019-02-22:
I agree @magland 
Can you do this?

magland, 2019-02-23:
@alejoe91. I will assign myself, but I'd like to wait until things stabilize some more before making the change. Ideally there would just be one place to make the change, rather than having instances of writing to binary files in various places throughout.

alejoe91, 2019-03-01:
@magland 

I'll try to modify spyking-circus to use .dat instead. Then we can just use the bindatrecordingextractor.

I think that the changes that you suggested can go in the numpy recording extractor only, and we have to make sure that everytime that we write npy we use that.

Do you agree?

colehurwitz, 2019-09-10:
Is this done?

closed 2019-09-25 by alejoe91