SpikeInterface/spikeextractors #88: separate out the extractors that cannot be verified via unit test

magland, 2019-02-28:
I suggest that we make a subdirectory in extractors for those that are in progress -- meaning that they are not well-covered by unit tests. Perhaps the directory should be called "devel" and we explain that those are under development and should be used at ones own risk.

If you all agree I can undertake this task.

alejoe91, 2019-02-28:
so I think we have to find a better way to test extractors that do not have (and won't have a writeRecording). I would leave them there now. the tests could pull data form zenodo and load them.

What do you think?

magland, 2019-02-28:
Okay, i guess they don't need to have a unit test. But we should at least believe that they work. Some of them may not be in that category. For example nwb (not quite sure if there are others).

alejoe91, 2019-02-28:
I think that for now we can comment out the nwb imports..
the other ones should be working

magland, 2019-02-28:
What I am proposing is that any extractor that may not be working (for example nwb) should be moved to a devel/ folder so that every extractor that's actually in the directory has our blessing, and is expected to work without much trouble.

alejoe91, 2019-02-28:
I do have files for OpenEphys, Intan, MEArec, Exdir. Should we have a minimal example on Zenodo or something similar? I believe @mhhennig has biocam. 

I'm ok in placing the nwbextracotr in a devel folder, or even make a dev branch for unfinished extractors

magland, 2019-02-28:
Sounds good. I think it would be very valuable having minimal examples of
each. Maybe first step is to put them on google drive or dropbox... unless
it is just as easy to put on zenodo (i haven't used it).

alejoe91, 2019-02-28:
I think @samuelgarcia has used Zenodo before? We can try that out :)

samuelgarcia, 2019-03-01:
For neo, we settle a "gin" repo to put our data here
https://web.gin.g-node.org/NeuralEnsemble/ephy_testing_data

Gin a tools based on git + gitannex that is able to track (in git sens) big files.
Gin platform is github like, easy to create an account and easy to join project.

I would strongly suggest to use this gin for storing this files rather than google drive or dropbox.
I allow contribution easily.

If you want you can even join this "ephy_testing_data" repo which is a collection of many format comonly used in ephy sphere.
To push data there you need an account for get data you can also use http so this is perfect for unittest.

magland, 2019-03-01:
Looks good to me.

colehurwitz, 2019-09-10:
Should we look into this gin stuff before or after the release? @samuelgarcia

samuelgarcia, 2019-09-10:
Before the release sounds to much work.
But lets leave this opened for later no ?

whitepine, 2020-08-20:
Hi there, I'm wondering if there's an example for each extractor as @magland  mentioned. I'm trying to get spikeinterface to read in an nwb file and am running into issues. Is nwb extractor in good working order? If so, an example of an nwb extractor working w/ spikeinterface would be great.

alejoe91, 2020-08-24:
Hi @jthermiz, is the problem fixed with the latest PR https://github.com/SpikeInterface/spikeextractors/pull/440?

jthermiz, 2020-08-28:
@alejoe91  @mhhennig  It looks like the problem has been resolved perhaps thanks #440