SpikeInterface/spikeextractors #91: OpenEphysRecordingExtractor.getChannelIds force to load everythin in memory

samuelgarcia, 2019-03-01:
OpenEphysRecordingExtractor.getChannelIds force to load everythin in memory

Here a bug I have in TridesclousSorter because saveProbeFile need channel_ids and getting channel ids read everything included multifly the gain.

See my bug:
and this issue https://github.com/CINPLA/pyopenephys/issues/11

```

 File "/home/samuel/Documents/projet/SpikeInterface/spiketoolkit/spiketoolkit/sorters/basesorter.py", line 77, in run
    self._setup_recording(recording, self.output_folders[i])
  File "/home/samuel/Documents/projet/SpikeInterface/spiketoolkit/spiketoolkit/sorters/tridesclous/tridesclous.py", line 47, in _setup_recording
    se.saveProbeFile(recording, probe_file, format='spyking_circus')
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/tools.py", line 177, in saveProbeFile
    _export_prb_file(recording, probe_file, format, radius=radius, dimensions=dimensions)
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/tools.py", line 305, in _export_prb_file
    if 'location' in recording.getChannelPropertyNames():
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/RecordingExtractor.py", line 275, in getChannelPropertyNames
    for channel_id in self.getChannelIds():
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/extractors/openephysextractors/openephysextractors.py", line 17, in getChannelIds
    return list(range(self._recording.analog_signals[0].signal.shape[0]))
  File "/home/samuel/Documents/projet/SpikeInterface/pyopenephys/pyopenephys/core.py", line 394, in analog_signals
    self._read_analog_signals()
  File "/home/samuel/Documents/projet/SpikeInterface/pyopenephys/pyopenephys/core.py", line 703, in _read_analog_signals
    signal=anas * gain,
MemoryError
```

alejoe91, 2019-03-01:
https://github.com/SpikeInterface/spikeextractors/pull/95/commits/0107703d5cb4abb723a87a57ed76f98350902fd7

alejoe91, 2019-03-01:
ca1d53c2be4125e4c90e4a0c4675d113e9c92e60

closed 2019-03-01 by alejoe91