SpikeInterface/spikeextractors #106: Removal of UnitSpikeFeatures

colehurwitz31, 2019-03-18:
I was considering removing UnitSpikeFeatures (set, get, etc.) from the SortingExtractor. Is anyone using this method extensively? 

We can already store arrays of values with the UnitProperty functions so I don't know if the UnitSpikeFeatures methods are necessary. Also, it is a bit annoying to keep track of multiple dictionaries. What are your guys thoughts?

@alejoe91 @magland @samuelgarcia @mhhennig

alejoe91, 2019-03-18:
Yes! I use it to load waveforms and PCA scores. Please keep it ;)

Spike features is very different than property, as each feature is for a spike, the property is for the unit.

colehurwitz31, 2019-03-18:
Ok, good to know :).

closed 2019-03-18 by colehurwitz31