SpikeInterface/spikeextractors #107: Property Handling in Curated Sorting Extractor

colehurwitz31, 2019-03-18:
Currently, in the CuratedSortingExtractor, properties are copied in from the parent_sorting. However, when two units are merged or a unit is split, the underlying properties are lost.

I think this is ok and something that makes sense given that there is no good way to split or merge properties (if one unit has a property and the other has the same property, can we really merge those properties?)

Thoughts?

@alejoe91 @magland

colehurwitz31, 2019-03-19:
I will implement SpikeFeature splitting and merging though as those should be a part of the CuratedSortingExtractor.

colehurwitz31, 2019-03-19:
I will make unit tests for CuratedSortingExtractor soon, however, if someone wants to test it out, that would be excellent. I plan to develop new postprocessing modules using the CuratedSortingExtractor as a base.

@alejoe91 @samuelgarcia

samuelgarcia, 2019-03-19:
I have not play with CuratedSortingExtractor yet, so I can't help here.
I am discovering the while module zone by zone.

closed 2019-03-20 by colehurwitz31