SpikeInterface/spikeextractors #111: Should MultiSortingExtractor only deal with non-continuous sorting extractors

colehurwitz31, 2019-03-19:
Currently, the MultiSortingExtractor can take either N continuous sorting extractors or N separate sorting extractors. Because of this, the current implementation is convoluted and some methods just don't work for the continuous case (features and properties).

Do you think we should remove the continuous case since it seems like it does not have much practical value and makes the code confusing and certain features difficult (impossible)?

@alejoe91 @samuelgarcia @magland @mhhennig

alejoe91, 2019-03-19:
Yeah the continuous case is tricky and I agree it wouldn't be used that much.
A better use case would be concatenating recordings and run sorting on all of them. In that case, though, we should be able to retrieve the start frame for each recording.

samuelgarcia, 2019-03-19:
I have not play with the MultiSortingExtractor, so I can't help.

colehurwitz31, 2019-03-19:
We can leave it in for now, but we maybe should consider removing it in the future

closed 2019-07-11 by colehurwitz