SpikeInterface/spikeextractors #128: Possible bug in MEArecRecordingExtractor

samuelgarcia, 2019-04-30:
I am try to make a notebook in spikeresult with file shared by Alesio.
This file are generated by MEArec.

Maybe I am doing something wring or maybe it is a bug or maybe a version dependency problem
WIth this 
```python
import spiketoolkit as st
import spikeextractors as se

mearec_filename = 'templates_100_SqMEA-10-15um_nc10_25-03-2019.h5'
rec0  = se.MEArecRecordingExtractor(mearec_filename)
gt_sorting0 = se.MEArecSortingExtractor(mearec_filename)
```

I get
```
Traceback (most recent call last):
  File "dev_debug_mearec_reader.py", line 14, in <module>
    rec0  = se.MEArecRecordingExtractor(mearec_filename)
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/extractors/mearecextractors/mearecextractors.py", line 31, in __init__
    self._initialize()
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/extractors/mearecextractors/mearecextractors.py", line 40, in _initialize
    self._fs = recgen.info['recordings']['fs']
```

@alejoe91  : Do you have any idea how to fix it ?

I did both pip install MEArec and clone from github.

samuelgarcia, 2019-04-30:
I am stupid.
This file is not a recording but template.
Sorry.

samuelgarcia, 2019-04-30:
Finally I have another bug...

```python
import spiketoolkit as st
import spikeextractors as se

p = './spikeresults/mearec_template/'

mearec_filename = p + 'recordings_50cells_SqMEA-10-15um_10.0_10.0uV_2.h5'
rec0  = se.MEArecRecordingExtractor(mearec_filename)
gt_sorting0 = se.MEArecSortingExtractor(mearec_filename)
```
```
Traceback (most recent call last):
  File "dev_debug_mearec_reader.py", line 8, in <module>
    rec0  = se.MEArecRecordingExtractor(mearec_filename)
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/extractors/mearecextractors/mearecextractors.py", line 31, in __init__
    self._initialize()
  File "/home/samuel/Documents/projet/SpikeInterface/spikeextractors/spikeextractors/extractors/mearecextractors/mearecextractors.py", line 39, in _initialize
    recgen = mr.load_recordings(recordings=self._recording_path, return_h5_objects=True)
  File "/home/samuel/Documents/projet/SpikeInterface/MEArec/MEArec/tools.py", line 218, in load_recordings
    info = load_dict_from_hdf5(F, 'info/')
  File "/home/samuel/Documents/projet/SpikeInterface/MEArec/MEArec/tools.py", line 429, in load_dict_from_hdf5
    return recursively_load_dict_contents_from_group(h5file, path)
  File "/home/samuel/Documents/projet/SpikeInterface/MEArec/MEArec/tools.py", line 448, in recursively_load_dict_contents_from_group
    for key, item in h5file[path].items():
AttributeError: 'Dataset' object has no attribute 'items'
```

colehurwitz31, 2019-04-30:
The MEArec extractor works with the newest files generated by MEArec, but is not backwards compatible with older recordings. When was the recording generated?

colehurwitz31, 2019-05-03:
Was this the issue? @samuelgarcia

alejoe91, 2019-05-03:
@samuelgarcia can you try the dev version?

samuelgarcia, 2019-05-03:
yep. We can close. With another recording it works.
The one was too old.

closed 2019-05-03 by samuelgarcia