SpikeInterface/spikeextractors #134: 3D geometry with MEArec extractor make sorter buggy

samuelgarcia, 2019-05-09:
I am try to run sorter on the recording generated by MEArec given Allessio in google drive.
The geometry appear to be 3D. I think this is a good option for the futur but in the actual state of sorter I think they all deal with 2D geometry.

So for instance, the PRB file converted look like this:
```
total_nb_channels = 100
radius = 100
channel_groups = {
     0: 
        {
           'channels': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99],
           'graph':  [],
           'geometry':  {
               0: [0.0, -67.5, -67.5],
               1: [0.0, -67.5, -52.5],
               2: [0.0, -67.5, -37.5],
               3: [0.0, -67.5, -22.5],
               4: [0.0, -67.5, -7.5],
               5: [0.0, -67.5, 7.5],
               6: [0.0, -67.5, 22.5],
               7: [0.0, -67.5, 37.5],
               8: [0.0, -67.5, 52.5],
```

and this prevent many sorters running.
I think that **_export_prb_file** should have an option  to force 2D and or raise error.

About the 3D geometry (wich is a good choice), it should be totally explicit for all extractor and not case by case.

alejoe91, 2019-05-09:
it does :) https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extraction_tools.py#L140
The `dimension` arg can be used. E.g in MEArec you would use `dimension=[1,2]`

This is done automatically by sorters which don't deal woth 3d locaitons (kilosort, spyking-circus) via the argument `electrode_dimensions`

samuelgarcia, 2019-05-09:
yes but **save_probe_file** is call inside each sorter blindly without the knowledge of wich dimenssion must be used or not.
This option **electrode_dimensions** is tricky (at least for beginners).

So in case of 3D (like MEArec extractor), some sorter should fail because of this.

**run_sorters** should work out-the-box without tricky args like this one.

We should prevent this kind of error with some magic dimenssion selection, at least for sorter based on PRB. Having 3D geometry is not in PRB specification, so we should not allow to write it.

no ?

samuelgarcia, 2019-05-09:
How many RecordingExtractor have implicitly the geometry ?
For theses extractor we should have option to load 2D geometry.
For instance, why did you choose to have 0.0 for all electrodes at dim=0 ?
Is dim 0,1,2 for x,y,z ?

alejoe91, 2019-05-31:
The `plane` information is present in MEArec. What about a `convert_to_2D` arg in the init (True by default) that loads the correct information?

closed 2019-06-08 by alejoe91