SpikeInterface/spikeextractors #141: Store spike train counts as properties

colehurwitz31, 2019-05-16:
I find myself extracting spike trains and getting there lengths all the times. Upon initialisation, do you think storing spike train counts automatically as a property and then giving an easy way to access them (get_unit_spiketrain_counts()) would be a good idea?

mhhennig, 2019-05-16:
Could take a while to compute for large data, but should be useful:
```python
self._num_spikes_units = [len(self.get_unit_spike_train(i)) for i in self.get_unit_ids()]
```

colehurwitz31, 2019-05-16:
I would store it in the properties dictionary, but I thought it would be useful as well

colehurwitz31, 2019-05-16:
We could have a Boolean flag in the sorting extractor parent class that is false by default for loading spike train counts

mhhennig, 2019-05-16:
Or create a function `get_unit_spiketrain_counts()` in the parent class that computes counts and adds them as property when first called? Could this potentially break anything?

colehurwitz31, 2019-05-16:
No, it wouldn’t break anything, but it should probably be optional

closed 2019-07-11 by colehurwitz