SpikeInterface/spikeextractors #142: MEArecRecordingExtractor and MEArecSortingExtractor are quite slow

samuelgarcia, 2019-05-22:
I am playing a bit with mearec theses days.
I found that MEArecRecordingExtractor and MEArecSortingExtractor are quite slow.
I don't have time to read the mearec code for IO but I think that **mr.load_recordings** load many thing in memory. there are no memmap or lazy loading.
Am I wrong ?

alejoe91, 2019-05-22:
There should be a memmap of the h5 files for the recording extractor. I will check.
For the sorting extractor it probably loads the waveforms in memory..we could delete the object later

samuelgarcia, 2019-05-22:
At the moment it is not a probleme of memory use but time to load everyhting.
I think you could put some load on demand for waveform (and everything).

If you had used the neo rawio API ...

alejoe91, 2019-05-22:
Eheheh we can discuss about that! Indeed spiketrains are neo spike trains, so it should be easy to load them with the neo api

samuelgarcia, 2019-05-22:
It is my secret plan : base spikeextractor on neo.rawio API with a simple wrapper.
But I don't think I will have before 2022. :)

colehurwitz, 2019-07-11:
All done with this @alejoe91 ?

alejoe91, 2019-07-11:
yep

closed 2019-07-11 by alejoe91

hoducha92, 2019-09-14:
Quá phức tạp dề nghị có tiếng 5nam

colehurwitz, 2019-09-14:
Not too complicated! Just making sure our file IO is up to speed :) if you have any questions, let us know.

hoducha92, 2019-09-14:
Đề nghị cho tiếng việt nam thì sẽ tiếp xúc nhiều người

colehurwitz, 2019-09-14:
Unfortunately, none of us speak Vietnamese. However, if you would like to help us translate our [readthedocs](https://github.com/SpikeInterface/spikeinterface/tree/master/doc) into Vietnamese, you can make a pull request. Here is a link about how to do this: https://docs.readthedocs.io/en/stable/localization.html. Thanks for checkout out our project!