SpikeInterface/spikeextractors #147: OpenEphysRecordingExtractor does not contain 'gain'

JoseGuzman, 2019-05-31:
I'm trying to perform spike sorting with spyking-circus from a binary file acquired with open-ephys. However, the gain object is missing. I checked it, and it is not present in the https://github.com/CINPLA/pyopenephys module.

```python
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-dc1fae000be4> in <module>()
----> 1 sorting_SC = st.sorters.run_spykingcircus(recording)

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/sorterlist.py in run_spykingcircus(*args, **kargs)
     84 
     85 def run_spykingcircus(*args, **kargs):
---> 86     return run_sorter('spykingcircus', *args, **kargs)
     87 
     88 def run_herdingspikes(*args, **kargs):

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/sorterlist.py in run_sorter(sorter_name_or_class, recording, output_folder, delete_output_folder, grouping_property, parallel, debug, **params)
     52                          parallel=parallel, debug=debug, delete_output_folder=delete_output_folder)
     53     sorter.set_params(**params)
---> 54     sorter.run()
     55     sortingextractor = sorter.get_result()
     56 

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/basesorter.py in run(self)
     87     def run(self):
     88         for i, recording in enumerate(self.recording_list):
---> 89             self._setup_recording(recording, self.output_folders[i])
     90 
     91         t0 = time.perf_counter()

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/spyking_circus/spyking_circus.py in _setup_recording(self, recording, output_folder)
     62         # save binary file
     63         file_name = 'recording'
---> 64         np.save(str(output_folder / file_name), recording.get_traces().astype('float32'))
     65 
     66         if p['detect_sign'] < 0:

~/git/spikeextractors/spikeextractors/extractors/openephysextractors/openephysextractors.py in get_traces(self, channel_ids, start_frame, end_frame)
     51         elif self._dtype == 'float':
     52             return self._recording.analog_signals[0].signal[channel_ids, start_frame:end_frame] * \
---> 53                    self._recording.analog_signals[0].gain
     54 
     55 

AttributeError: 'AnalogSignal' object has no attribute 'gain'
```

alejoe91, 2019-05-31:
Hi,

In which format did you save the open ephys recording? binary or openephys?
I haven't tested properly with the openephys format, which will be discontinued soon, but I'll double check.

The gain is part of the AnalogSignal class (https://github.com/CINPLA/pyopenephys/blob/master/pyopenephys/core.py#L35)

Cheers
Alessio

alejoe91, 2019-05-31:
Forgot to make a new release!!! One sec :)

JoseGuzman, 2019-05-31:
It's a binary file format.

The pyopenephys version is '1.0.5'

alejoe91, 2019-05-31:
exactly..I had version 1.0.6 on setup.py but I forgot to tag it. Should be up in Pypi now (https://pypi.org/project/pyopenephys/#history)

Can you try to reinstalling it and run it again?

Thank for spotting it!
Alessio

alejoe91, 2019-05-31:
Let us know if you have other issues :) We use open ephys both for tetrode recording and silicon probes, so we have several probe files and pipelines.
If you are interested this is how we preprocess, sort, and save to phy:
https://github.com/CINPLA/expipe-plugin-cinpla/blob/dev/expipe_plugin_cinpla/scripts/openephys.py#L114-L271

JoseGuzman, 2019-05-31:
The grain object is present after `pip install pyopenephys --upgrade # from 1.0.5 to 1.0.6 `. Note that you are using spyking-circus version 0.7.6 (current is 0.8 and cannot be upgraded with pip)

now, when I prepare the sorter
```python
sorting_SC = st.sorters.run_spykingcircus(recording =myrec, probe_path ='../spyking-circus/ASSY-77-P4.prb')
```
I got the following error:
```python
.dat:  continuous.dat
Estimated samples:  44364544  Numchan:  67
'location' property is not available and it will not be saved.
'group' property is not available and it will not be saved.
--------------------------------------------------------------------------

Traceback (most recent call last):

[[19882,1],0]: A high-performance Open MPI point-to-point messaging module

  File "/home/segundo.martinez/miniconda3/envs/python3/bin/spyking-circus", line 10, in <module>

was unable to find any relevant network interfaces:

    sys.exit(main())

  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/circus/scripts/launch.py", line 350, in main

Module: OpenFabrics (openib)

    n_edges = get_averaged_n_edges(params)

  Host: pcl-imba-3

  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/circus/shared/probes.py", line 79, in get_averaged_n_edges

    nodes, edges = get_nodes_and_edges(parser)

Another transport will be used instead, although this may result in

  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/circus/shared/probes.py", line 72, in get_nodes_and_edges

lower performance.

    edges[i] = get_edges(i, parser.probe['channel_groups'][key])

  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/circus/shared/probes.py", line 63, in get_edges

NOTE: You can disable this warning by setting the MCA parameter

    pos_x, pos_y = channel_groups['geometry'][i]

btl_base_warn_component_unused to 0.

KeyError: 'geometry'

--------------------------------------------------------------------------

##################################################################

#####           Welcome to the SpyKING CIRCUS (0.7.6)        #####

#####                                                        #####

#####              Written by P.Yger and O.Marre             #####

##################################################################

File          : /home/segundo.martinez/SiliconProbes/spyking-circus/test_spykingcircus/recording.npy

Steps         : filtering, whitening, clustering, fitting

Number of CPU : 12/24

Parallel HDF5 : False

Shared memory : True

Hostfile      : /home/segundo.martinez/spyking-circus/circus.hosts

##################################################################

-------------------------  Informations  -------------------------

| Number of recorded channels : 67

| Number of analyzed channels : 67

| File format                 : NUMPY

| Data type                   : float32

| Sampling rate               : 30 kHz

| Duration of the recording   : 24 min 38 s 818 ms

| Width of the templates      : 3 ms

| Spatial radius considered   : 100 um

| Threshold crossing          : negative

------------------------------------------------------------------

-------------------------  Informations  -------------------------

| Using only 12 out of 24 local CPUs available (-c to change)

------------------------------------------------------------------

---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
<ipython-input-3-52901809e19f> in <module>()
----> 1 sorting_SC = st.sorters.run_spykingcircus(recording =myrec, probe_path ='../spyking-circus/ASSY-77-P4.prb')

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/sorterlist.py in run_spykingcircus(*args, **kargs)
     84 
     85 def run_spykingcircus(*args, **kargs):
---> 86     return run_sorter('spykingcircus', *args, **kargs)
     87 
     88 def run_herdingspikes(*args, **kargs):

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/sorterlist.py in run_sorter(sorter_name_or_class, recording, output_folder, delete_output_folder, grouping_property, parallel, debug, **params)
     52                          parallel=parallel, debug=debug, delete_output_folder=delete_output_folder)
     53     sorter.set_params(**params)
---> 54     sorter.run()
     55     sortingextractor = sorter.get_result()
     56 

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/basesorter.py in run(self)
     93         if not self.parallel:
     94             for i, recording in enumerate(self.recording_list):
---> 95                 self._run(recording, self.output_folders[i])
     96         else:
     97             # run in threads

~/miniconda3/envs/python3/lib/python3.5/site-packages/spiketoolkit/sorters/spyking_circus/spyking_circus.py in _run(self, recording, output_folder)
     98         retcode = _run_command_and_print_output(cmd)
     99         if retcode != 0:
--> 100             raise Exception('Spyking circus returned a non-zero exit code')
    101         if self.params['merge_spikes']:
    102             if self.debug:

Exception: Spyking circus returned a non-zero exit code
```

alejoe91, 2019-05-31:
Can you paste the probe file here?

alejoe91, 2019-05-31:
Seems like it's not loaded correctly. It should add a 'location' property to each channel, but the warning message that you get says that 'location' property is not available.

JoseGuzman, 2019-05-31:
> Can you paste the probe file here?
```python

"""
ASSY-77-P4.prb

Author: Jose Guzman, jose.guzman@guzman-lab.com
Created: Mon May  6 12:22:42 CEST 2019

The probe file for the silicon probe model ASSY-77-P4
from Cambridge NeuroTech. The probe consists of FOUR 250 um-separated 
shanks of 16 recording sites. The 16 recording sites cover a distance of 
200 um and are spaced every 25 um. The total number of electrodes
is 64.

The file creates a main python dictionary called channel_groups. The 
dictionary contains four dictionaries called '0', '1', '2' and '3'
corresponding to the four schanks of the silicon probe (A,B,C and D). 
Every shank contains another dictionary with "channels", "graph" and 
"geometry" having the channel number.

"""

#=========================================================================
# Skyking-circus parameters
#=========================================================================
total_nb_channels = 64 # number of recording sites
radius = 250 # spatial extent (in um)
channel_groups = {1: dict(), 2: dict(), 3: dict(), 4: dict()} # four shanks

#=========================================================================
# "channels" defined from channel mapping
#=========================================================================
shankA = [48, 37, 33, 39, 35, 41, 47, 44, 34, 42, 32, 43, 45, 40, 38, 36] 
shankB = [53, 62, 51, 60, 49, 59, 46, 57, 50, 55, 52, 54, 56, 58, 61, 63]
shankC = [11,  0, 13,  2, 15,  5, 16,  7, 12,  9, 10,  8,  6,  4,  3,  1] 
shankD = [14, 27, 31, 25, 29, 23, 17, 18, 28, 20, 30, 21, 19, 22, 24, 26]

#=========================================================================
# Assign "channels" from channel maps 
#=========================================================================
channel_groups[1]["channels"] = shankA
channel_groups[2]["channels"] = shankB
channel_groups[3]["channels"] = shankC
channel_groups[4]["channels"] = shankD

#=========================================================================
# Assign "geometry" for calculated locations 
#=========================================================================
def get_locations(xloc = 0, step = 25):
    """
    generate a list of  16 (x,y) coordinates (in um) for a shank

    Arguments:
    ----------
    initloc (int): the starting xlocation of the first
        recording site up in the probe. After that, locations move
        to the left one step, and down step. Zero by default.
    step (float): the x-y distance between electrodes. Default 25 um.

    Return:
    -------
    A list with (x,y) coordinates for 16 electrodes in the shank

    Example:
    --------
    get_geometry(xloc = 250, step = 25) # return coord shank B
    """

    mycoor = list()
    for yloc in range(0, 8*step, step):
        right = (xloc, yloc) # rigth electrode
        left  = (xloc-step, -(yloc+step) ) # left-down electrode

        mycoor.append(right)
        mycoor.append(left)
    
    return mycoor

# create dictionaries 'zipping' shanks and locations
channel_groups[1]["geometry"] = dict( zip(shankA, get_locations(0,  25)))
channel_groups[2]["geometry"] = dict( zip(shankB, get_locations(250,25)))
channel_groups[3]["geometry"] = dict( zip(shankC, get_locations(500,25)))
channel_groups[4]["geometry"] = dict( zip(shankD, get_locations(750,25)))
    

#=========================================================================
# Assign "graph" of adjancenct nodes to have phy compatibility
#=========================================================================
def get_graph(shank):
    """
    generate adjacency graph. It adds the two closest electrodes for 
    for every electrode in the shank.
    
    Arguments:
    ---------
    A shank with the list of electrodes

    Returns:
    --------
    A list tuples with the two closest electrodes
    It is 2x the size of the shank list.
    """
    mygraph = list()
    i = 0;
    while i<len(shank)-2:
        mygraph.append( (shank[i],shank[i+1]) ) 
        mygraph.append( (shank[i],shank[i+2]) )
        i +=1

    return mygraph
    
channel_groups[1]["graph"] = get_graph(shankA)
channel_groups[2]["graph"] = get_graph(shankB)
channel_groups[3]["graph"] = get_graph(shankC)
channel_groups[4]["graph"] = get_graph(shankD)
```

JoseGuzman, 2019-05-31:
> Seems like it's not loaded correctly. It should add a 'location' property to each channel, but the warning message that you get says that 'location' property is not available.

Yes, the duration of the recording is different if I load it with spyking-circus (`| Duration of the recording   : 6 min 27 s 34 ms`) or with pyopenephys (`| Duration of the recording   : 24 min 38 s 818 ms`)

alejoe91, 2019-05-31:
Is the recording in openephys 24 or 6 minutes??? 

Regarding the probe file, currently we support a simple and universal definition. I'll try out with your probe file, modify it, and paste it back :)

JoseGuzman, 2019-05-31:
the recording is exactly 24 min 38 s 818 ms as prompted! :)

alejoe91, 2019-05-31:
great!

So the probe file looks good actually. Can you try to do this instead of passing the probe file?
Note that in our design, the probefile only provides group and location information. You can pass the other information (adjacency_radius) when you call the sorter:

```
import spikeextractors as se
import spiketoolkit as st

myrec = se.OpenEphysRecordingExtractor('path-to-oe-folder')
myrec = se.load_probe_file(myrec, ''../spyking-circus/ASSY-77-P4.prb'')
sorting_SC = st.sorters.run_spykingcircus(recording =myrec, probe_path ='../spyking-circus/ASSY-77-P4.prb', adjacency_radius=250)
```
The `get_graph` is implemented internally.

Let me know if it works! It works locally on my machine on simulated 64-chan data.

I'll open an issue for documenting the correct use of the probe file

alejoe91, 2019-05-31:
@JoseGuzman just noticed that you used `probe_path` instead of `probe_file`! `probe_file` is the correct argument!!!

alejoe91, 2019-05-31:
You will still need to pass the `adjacency_radius` separately

JoseGuzman, 2019-05-31:
It works now, with an unsual number of spikes! Very strange!
```python
Here comes the SpyKING CIRCUS using 12 CPUs and 72 templates...

Gathering data from 12 nodes...

-------------------------  Informations  -------------------------

| Number of spikes fitted : 132359

------------------------------------------------------------------

/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/h5py/_hl/dataset.py:313: H5pyDeprecationWarning: dataset.value has been deprecated. Use dataset[()] instead.
  "Use dataset[()] instead.", H5pyDeprecationWarning)
```

> You will still need to pass the `adjacency_radius` separately

Great, then the whole conditions in *params will be specified as arguments. Looking forward to the documentation!!!!

closed 2019-05-31 by JoseGuzman

eejd, 2021-02-23:
I am having the same problem (missing 'gain') but am using the open ephys file format (I have a lot of tetrode data recorded this way, so I need to access this file format). Is this problem caused by the file format? I see it set to 'fixed_gain' in the pyopenephys module (in theory). The data loads, but I get this problem when I try to access it (e.g. using  plot_timeseries). 

pip installed versions:
pyopenephys         1.1.3
spikecomparison     0.3.1
spikeextractors     0.9.3
spikefeatures       0.1.1
spikeinterface      0.11.0
spikemetrics        0.2.3
spikesorters        0.4.3
spiketoolkit        0.7.2
spikewidgets        0.5.1

alejoe91, 2021-02-24:
Hi @eejd 

We are in the middle of updating packages. Installing the previous version of pyopebephys should fix it for now:
`pip install pyopenephys==1.0.7`

eejd, 2021-02-24:
Ok, I see that now the pyopenephys is storing as .gains for each channel. Also, have you considered (or have an opinion) on using pyopenephys vs the open-ephys-tools (from open ephys) for the base loading? I'll revert the openephys version now and confirm it's working.

alejoe91, 2021-02-24:
They both work :) with the new pyopenephys version (which is currently supported by the master branch -- that will be released soon), we have continuous integration with test files. 

In general, we are trying to move more and more towards NEO-based extractors. As soon as a NEO wrapper is in place for the openephys binary format, we'll probably make the switch.

eejd, 2021-02-24:
Thanks for the fast responses. I can confirm that the reversion to 1.0.7 seems to have solved the problem. I have a related question, which is why when I am selecting a single channel (I believe) in the extractor via (..., recording_id=0), it is still loading all the channels from the session folder. Is this a feature of the pyopenephys loader or the OpenEphysRecordingExtractor? I would like to be able to load up individual tetrodes (e.g only load into memory 4 channels at a time). Sorry, if this isn't the appropriate place for this question, I can move it.

alejoe91, 2021-02-24:
The extractor will memory map all data. If you want to only select a tetrode, you can do so using the `SubRecordingEctractor`:

`recording_tetrode_0 = se.SubRecordingExtractor(recording, channel_ids=[0, 1, 2, 3])`