SpikeInterface/spikeextractors #149: make read the docs

alejoe91, 2019-06-02:
- recording extractor
- sorting extractor
- available extractors
- using a probe file
- (curating results)

colehurwitz, 2019-06-02:
I will look into doing this soon

alejoe91, 2019-06-02:
I've started :) will make a PR later, than we can work on that

colehurwitz, 2019-06-02:
Oh, nice :). Sounds good!

samuelgarcia, 2019-06-02:
@alejoe91 : I experienced difficulties in the past with read the doc and auto build because of dependencies. Maybe it is easier today. I would advice to try something simple at the begenning and see how it build. Then you can make more complexe. For instance with sphink-galery plugin (to avoid notebook in git).

alejoe91, 2019-06-12:
@colehurwitz do you have time to fill in the missing parts? :)

colehurwitz, 2019-06-12:
Yes, I can get to those in the near future!

alejoe91, 2019-06-30:
Still missing the probe file handling

alejoe91, 2019-06-30:
but there's another issue open

closed 2019-06-30 by alejoe91