SpikeInterface/spikeextractors #156: Kernel issue with spikeextractors

vncntprvst, 2019-06-07:
Hi,
I have the basic issue that when importing spikeextractors, the kernel crashes.
The jupyter notebook displays a pop up window with this:
> Kernel Restarting
> The kernel appears to have died. It will restart automatically.
This happens in Ipython too (minus the message).
Same thing with spiketoolkit ... 
I'm testing this on a Windows 10 machine, running Python 3.7.0. 
Any idea what's going on? 
Thanks

alejoe91, 2019-06-07:
Hi, I have never seen it.. can you try to import it in Python and also to try python3.6? I use SpikeInterface on windows 10 python3.6 with no problems.

What's in your environment? Do you use anaconda? If not, I highly recommend to use it :)

vncntprvst, 2019-06-07:
I do use Anaconda, and indeed Python 3.7 was the issue. Import works fine using a 3.6 kernel. 
Thanks

closed 2019-06-07 by vncntprvst