SpikeInterface/spikeextractors #169: Add sampling frequency to SortingExtractors that support it

colehurwitz, 2019-06-12:
I was thinking an optional sampling_frequency argument can be added to sorting extractors that do not have access to the sampling rate:

MDASortingExtractor(firings, sampling_frequency=None)

colehurwitz, 2019-06-12:
@samuelgarcia  Is the .npz sampling frequency code ok? If so, I can uncomment it and commit it

samuelgarcia, 2019-06-12:
I think it is better to have explicitly self._sampling_frequency = None for theses format.

The only consequence is that the user will have to provide manually for SpikeComparison otherwise it will raire an error.

About npz you can uncomments but it has never been tested.
So please add 2 line in the test_nzp....

colehurwitz, 2019-06-12:
I can add that as a test!

colehurwitz, 2019-06-12:
Just realized a bit of a problem... numpy sorting extractors don't support sampling frequency and we use it in our tests to write to different formats.

colehurwitz, 2019-06-12:
Should we add that to NumpySortingExtractor as an argument that is None by default?

samuelgarcia, 2019-06-12:
Yes.

closed 2019-06-17 by alejoe91