SpikeInterface/spikeextractors #170: NEO extractors

colehurwitz, 2019-06-12:
Does making NEO extractors make sense or is that not something worth doing? @samuelgarcia

samuelgarcia, 2019-06-12:
Of course we have to do it :)
In my mind it was a more long term project.

neo and spikeextractor have a super strong overlap. This will lead to energy disperssion.
So a rapprochement between se and neo should be good. And it should be more than an extra tor based an neo.

colehurwitz, 2019-06-12:
Yes, that is a good idea. We can come up with more ideas in a meeting at some point.

samuelgarcia, 2019-10-25:
Soon. Soon.
I am totally busy on other project but I will be back.
:)

McD-NMI, 2019-10-29:
I was reading the Neo documentation recently though, and, it seems like the NeoHdf5IO module is only for reading from older Neo H5 files, and for writing they suggest using NIXIO module. I'm not sure if the plan was to make a Neo H5 extractor... or an extractor that can make use of any of the NeoIO modules? It seems that perhaps Neo data structure is beign replaced by Nix?

samuelgarcia, 2019-10-29:
NeoHdf5IO is old. Its was kept for backward compatibility.

Some time ago Andrew Davison planed to make a new NeoHdf5IO with neo object model that would have both reading/writting ability and that would be a very simple transcription into HDF5 of the neo tree Block>Segment>(AnalogSignal, SpikeTrain, Event, Epoch) with all annotation and array_annotation features.
I don't think he will have time to do it soon.

closed 2020-04-22 by alejoe91