SpikeInterface/spikeextractors #180: HS2 extractor is extremely slow

alejoe91, 2019-06-17:
tested on the retina dataset, it takes 20 minutes to load a sorting extractor

samuelgarcia, 2019-06-17:
And mearec extractor is "quite" slow :)

samuelgarcia, 2019-06-17:
npz is quite fast :)

alejoe91, 2019-06-17:
I think MEArec is quite slow because it loads waveforms..I'll make it optional

samuelgarcia, 2019-06-17:
And it load signals also, no ?

alejoe91, 2019-06-17:
no all memmapped

colehurwitz, 2019-06-17:
Will work on this. Probably has to do with all the extra info we load in as properties. We could always make this optional if that is the problem.

mhhennig, 2019-06-19:
I checked, it is `set_unit_spike_features` that's rather slow. I think it is all the checking it does. It should be possible to circumvent this function during creation, correct?

mhhennig, 2019-06-19:
Just made a PR with some changes that speeds up the loading of extra info in HS2 files.

alejoe91, 2019-06-20:
can we close this?

colehurwitz, 2019-06-20:
Yep!

closed 2019-06-20 by colehurwitz