SpikeInterface/spikeextractors #204: static typing

magland, 2019-07-02:
I have been using static typing in a subset of spikeforest (and plan to put it in everywhere). See
https://github.com/flatironinstitute/spikeforest/blob/master/mountaintools/mountainclient/mountainclient.py

@colehurwitz, I know that you mentioned this before

This, combined with something like mypi checker or the awesome pyright vscode plugin goes a long way to catch type errors before they cause exceptions. Eliminates a lot of bugs and makes usage much more clear. For example, solves the problem of the loose usage of str vs Path objects (@alejoe91).

I propose static typing all of SpikeInterface. I can take the lead if you guys all agree.

@samuelgarcia @alejoe91 @colehurwitz

samuelgarcia, 2019-07-02:
I am OK with the static typing.
But we need to make this steps by steps with a clear roadmap.
The very best for this would be to save a date and to make in parallel each one a module or submodule (soters, comparison, preprocessing, ...) in short time window.
@magland you would be the coordinator.
This kind of big change could colide with other changes, it has to be done quite fast. (cf camelcase change)

Before this, I would happy to finish the SF/ST integration. (sorter+mertics)

magland, 2019-07-02:
Okay, we can work on finishing integration first. Static typing is very nice in python in that you can have it in a subset of the files without it causing problems. So I might start including it in some of the newer contributions, as you have seen already.