SpikeInterface/spikeextractors #214: Need a way to serialize extractor.

samuelgarcia, 2019-07-16:
@colehurwitz @alejoe91 @magland 

We need a standard and simple way to "serialize" our extractor (spike and recordings).

  * One need is the Cole GUI
  * Another could be the launcher. because now launcher take a list of recordings. If we want to 
    put it in a queue for distributed computing we need to serialize.
 
The most obvisous way is to add methods in **RecordingExtractor** and **SortingExtractor**:
  * to_bytes() or dump()
  * from_bytes() or load()

It would be similar to **\_\_get_state\_\_** of the pickle module.

Theses methods would be optional. In short not all extractor would have dump/load ablility.

For instance **BinDatRecordingExtractor.dump()** could like something like this:

```python
class BinDatRecordingExtractor():

   def dump(self):
       d = {'class' : 'BinDatRecordingExtractor',
              'kargs' : {
                    'dafile' : '...',
                    'samplerate' : self.get_sampliong_rate(),
                    'dtype' : self._dtype,
                    'numchan' : self._numchan,
                    'frames_first' : self._frames_first,
              }
          }
    return json.dumps(d)

    @classmethod(cls)
    def load(cls, s):
        d = json.load(s)
        return cls(**d)
```

Some question:

  * wich technology json? yaml ? pickle ?
  * what do you think ?
  * how to deal with ectractor that have an extractor as input ? could allow also a serialize form?
    (that would lead to srialize chain)

@colehurwitz  : i don't known and don't have look yet in spikely how you did that. i will do.

colehurwitz, 2019-07-16:
What I plan to do for the GUI is to use the extractorlist.recording_extractor_dict or extractorlist.sorting_extractor_dict. Then I can just store the name and the params and instantiate it later (doesn’t really work for loaded properties, but that is part of our pipeline anyways)

samuelgarcia, 2019-07-16:
OK you store params before the object is created.

The scenario I propose is to get params when the object is already created, the 2 things could converge but we would need that al params would be store stored as attribute without name modification (dtype -> _dtype).

colehurwitz, 2019-07-16:
Ok, that works as well. Yeah, we were planning on using the name and pre-stored parameters, but serializing works too.

samuelgarcia, 2019-07-18:
@magland @alejoe91 : any comments ?

samuelgarcia, 2020-01-22:
@alejoe91 : any comments ?

samuelgarcia, 2020-01-22:
ok but put neo before this :)

alejoe91, 2020-01-29:
I'll make a draft PR for this very soon with a minimal example

alejoe91, 2020-03-31:
implemented

closed 2020-03-31 by alejoe91