SpikeInterface/spikeextractors #223: Complete NWB extractors, including write function

mhhennig, 2019-08-06:
Requirements:
- read raw data with a `RecordingExtractor` (partly done)
- read sorted data with a new `SortingExtractor`
- write functions for both
- allow writing of recording and sorting to the *same* file

luiztauffer, 2019-08-08:
Hi @mhhennig ! =)
We’re doing exactly this in our fork of the project: https://github.com/ben-dichter-consulting/spikeextractors/blob/master/spikeextractors/extractors/nwbextractors/nwbextractors.py 
comments and suggestions are welcome! We might PR soon

mhhennig, 2019-08-08:
Hi @luiztauffer, nice to see you again here!

Any questions ask here, I'll assign @colehurwitz, who conceived the extractors.

Have you thought about metadata? We have deliberately left any of this out of the extractors as we wanted light-weight access to data relevant to sorting, and nothing else. I have been thinking it may be good to just pass that through, without any parsing or similar. Any ideas?

bendichter, 2019-08-08:
Hey @mhhennig and @colehurwitz 

I was thinking we could handle metadata with optional dict args called e.g. `nwbfile_kwargs` which would, if defined, pass arguments to the `NWBFile` constructor. No functionality would depend on this, so users could continue to use this as a simple converter of just the essential data or could add metadata if they want to. Do you think that would work?

colehurwitz, 2019-08-08:
I think this should work! Having it as an optional argument to this specific extractor means that other extractors won’t be affected by these changes. 

Btw, how do you handle writing other files that don’t have a lot of metadata into NWB? If I wanted to write the .mda format from mountainsort into NWB, can your sorting extractor handle that well even though there isn’t metadata?

bendichter, 2019-08-09:
I'm not familiar with that particular format. A lot of the NWB meta data
fields are optional and for the ones that are not we can put place holders
in if we don't have the information.

closed 2019-10-17 by colehurwitz