SpikeInterface/spikeextractors #225: request: NeuralynxExtractor

bendichter, 2019-08-06:


samuelgarcia, 2019-08-26:
This will be handle when I will implement the neo extractor layer.

bendichter, 2019-09-25:
@samuelgarcia We are trying to use SpikeExtractors as core for much of our conversion framework, and this is a critical piece. What's the timeline on this?

alejoe91, 2019-09-25:
@bendichter @samuelgarcia maybe for now we can have a general NeoExtractor that parsed any neo io to a si extractor. This should be quite quick to implement

samuelgarcia, 2019-09-25:
For convertion stuff you should go for neo directly I think...
neo is more stable and have more feature (multi segment, multi sample rate, annoation, array_annoations, lazy loading...)

About the neo layer inside spikeextractors, I can't promise deadline, I hope I will start this this autumn. We have a lot of stuf  in neo, we are about to release the 0.8.

After the relase I plan to start the layer a bit. I can start with this neuralynx but the layer will be stable after more IO because it will have specific case.

samuelgarcia, 2019-09-25:
@alejoe91 : I would prefer to make something strong more something quick.
My final goal is to move all readr of extractor inside neo. And each extractor would be a thin class that wrapp another neo class.
The multi segment/multi sampling rate/ have to be deal carrfully, in short in that case we must have a system of option to select the desired analogsignal when there are several.

bendichter, 2019-09-25:
It looks like neo's `neuralynxrawio.py` only supports reading neuralyx, not writing. Is there a plan for supporting write?

samuelgarcia, 2019-09-25:
No writting to proprietary format is hudge, dangerous and useless task ...
Reading is already super difficult and specifically to maintain because all theses format evolve in times with versions...

alejoe91, 2019-09-28:
@bendichter for now we could maybe use the neo IO as a template to make a RecordingExtractor. that should be relatively quick

alejoe91, 2020-03-31:
implemented

closed 2020-03-31 by alejoe91