SpikeInterface/spikeextractors #227: units of measurement / magnitude

bendichter, 2019-08-09:
It is common for data acquisition formats to store voltage data as `int16` values and have a scaling factor to convert these values to volts (e.g. SpikeGLX). It looks like spikeextractor ignores these scaling factor values and returns data in an arbitrary magnitude. As a result, the units of measurement are lost during conversion from one data format to another. Would it be possible to include `get_scaling_factor` or something so that we can resolve data in volts?

alejoe91, 2019-08-10:
@bendichter I think this should be handled by each extractor. I implemented it in the `OpenEphysRecordingExtractor` and you can get traces eithe in int16 or in float (uV) (https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/openephysextractors/openephysextractors.py#L42-L53). I agree we should add something like that in the SpikeGLXExtractor.

Also, the `get_traces` function should allow to return traces both in `int16` or `uV`, as many sorters use `int16` directly.

We could have both a `get_scaling_factor`  and  a `get_dtype`. In case the traces are in uV, then the scaling factor could be one. 

@colehurwitz @bendichter what do you guys think?

bendichter, 2019-08-10:
@alejoe91 Thanks for pointing out that feature. NWB allows you to have it both ways: store data in `int16` which saves memory and also a conversion factor which allows you to convert to volts (much like the `gain` value of OpenEphys). That's why I thought a `get_scaling_factor` would be particularly useful for us. Is it specified somewhere that the data is converted to uV as opposed to mV or V? If not, can we do that?

@luiztauffer found a SpikeGLX library that converts to volts (https://billkarsh.github.io/SpikeGLX/Support/SpikeGLX_Datafile_Tools.zip), so we can base it off of that.

alejoe91, 2019-08-10:
@bendichter I'd prefer using uV as spike amplitudes are not e intuitive imo. 
We can add a `return_unit` in the `get_traces` from the base class and we can convert to uV, mV, or V in the get_traces on the fly. How does it sound? If return_unit is None and the file is binary then int16 is returned. We can definitely port some code from the SpikeGLX library. 
I'm a bit busy next week, would you have time to give it a try?

bendichter, 2019-08-11:
@alejoe91 Having a `get_scaling_factor` entirely solves our problem. Optionally applying the conversion in `get_traces` sounds find to me, but I probably wouldn't use it when converting to NWB:N since I'd prefer to store the data as type `int16` and keep the `scaling_factor`. But one thing I really need is to make sure we are consistent with what units `scaling_factor` converts to. The NWB conversion factors get you to SI base units (V), but if you want `uV` that's fine with me. It's trivial to convert from there as long as we are consistent.

alejoe91, 2019-08-13:
Ok then. I would prefer using `uV` as standard. You can convert in the extractors. Would you be able to implement that? Otherwise we can do it next week :)

luiztauffer, 2019-08-16:
I have datasets from SpikeGLX saved in different format pairs, example:

filename.imec0.ap.bin
filename.imec0.ap.meta

filename.imec0.lf.bin
filename.imec0.lf.meta

filename.nidq.bin
filename.nidq.meta

For the first pair (ap) SpikeGLXRecordingExtractor works fine. For the other pairs, no. This is caused by not reading the metafiles correctly.

[SpikeGLX](https://billkarsh.github.io/SpikeGLX/#offline-analysis-tools) provides a set of functions for easy handling of these files in python. They account for these files diversity and also other things, such as recovering channel gains. 

Using the functions they provide, I could read the important metadata in my files and extract the data correctly. I thought it would be a good idea to incorporate their reading functions to SpikeGLXRecordingExtractor. #228 
Let me know what you think or if you have suggestions to change it.

colehurwitz, 2019-08-23:
#231

closed 2019-08-23 by colehurwitz

samuelgarcia, 2019-08-26:
Hi all.
I am happy with this. Thank you.
I would also implement "get_dtype" very soon.

colehurwitz, 2019-08-26:
I implemented this using channel properties in #231 so we didn't have to change anything in the API . Good to see you back Samuel! I hope you had a nice vacation. :)