SpikeInterface/spikeextractors #232: Proposal: using contexts

bendichter, 2019-08-25:
I would like to propose that we support contexts as an optional workflow. 

For some filetypes it is best practice to close them after use. The NwbExtractors currently handle this by opening and closing the file with every method call, but this is inefficient. We could imply add a `close` method, but this would make the API different between extractors. Also, contexts are a preferred way to manage opening and closing files because the close will be called in the case of an error within the context. In the case of extractors that do not require a close method, the exit function of the context would just be `pass`. The syntax would look like this:

```python
with SpikeGlxExractor('fpath') as re:
    data = re.get_traces()
    ...
```
I think it will be possible to support this feature without changing the current form of the API at all. The API in its current form will continue to work, and this would be added as another way to use the API.

colehurwitz, 2019-08-26:
This is very interesting. I like the idea. I am not sure how to implement it myself, but if it isn't too hard, I think it would be a nice addition.

colehurwitz, 2019-08-26:
What are your thoughts @magland @alejoe91 @samuelgarcia ?

samuelgarcia, 2019-08-26:
The idea is cool but we need many internal modification since many extractor do a memap in the init...

bendichter, 2019-08-26:
I'm glad you guys like the idea. Through the magic of inheritance I really
don't think it will be much work

mhhennig, 2019-08-28:
Nice, this could solve a problem with h5py I've had with hdf5 files not getting closed. Is it correct that an extractor would then need a method `__exit__()`, which takes care of open files?

alejoe91, 2019-08-29:
@bendichter would you implement the `__enter__` and `__exit__` in the base extractors and just have `pass`. So other extractors with file IO can override it but the context notaiton will work for all. 

Does the context enforce to use the `with` notation? can you also add a destructor and close the file there in case one prefers not to use the context? that should solve the file closure as well.

bendichter, 2019-08-29:
@alejoe91 you would be able to continue to use the current syntax. It does not enforce this new notation. `__enter__` would call `__init__` and `__exit__` would just be `pass`