SpikeInterface/spikeextractors #233: transpose breaks lazy read

bendichter, 2019-08-25:
The recording files are often too large to fit into memory all at once, so it is important for us to support lazy read. For NWB (and perhaps other supported formats) this is not currently possible because we are required to transpose the data:

https://github.com/SpikeInterface/spikeextractors/blob/d90dd8e2402d516d2d69febf17de3648ae7cb97d/spikeextractors/extractors/nwbextractors/nwbextractors.py#L60

To solve this problem, I think we need to employ a lazy transpose method. This will provide a data view that is the transpose of the original data view, which will allow us to return a memory map and not the data itself. From there, we can iterate over time or channels and read/write in a piece-wise fashion. I am not aware of any implementation of this, but I suspect there is something out there. Maybe xarray or dask could help?

bendichter, 2019-08-25:
@luiztauffer

colehurwitz, 2019-08-26:
Agreed, this is important. It seems like xarray could work for dask-backed DataArrays, but not for numpy-backed DataArrays. 

http://xarray.pydata.org/en/stable/generated/xarray.DataArray.transpose.html

samuelgarcia, 2019-08-26:
Hi @bendichter.
The problem is the copy here not the transpose. No ?
In numpy the transpoe is lazy because it only affect the strides.

What happen if 
```data = ts.data.T ```

mhhennig, 2019-08-28:
Is it perhaps possible to first slice and then transpose? Slicing creates a view, so transposing that should be fast(er) than the whole chunk. Haven't tried yet though...

bendichter, 2019-08-29:
if we want start_frame and end_frame to work with lazy read we would also need to implement lazy slicing- something like [this](https://gist.github.com/stuarteberg/7ecd8cb7b24d12f4ffd9)

```python
class SimpleView(object):
    """
    Simple view-like object of an array-like object.
    Slices taken with __getitem__ are combined and stored for later use, 
      but the actual data is not extracted from the underlying array-like 
      object until __array__ is called.
      
    Warnings about the current implementation: 
    - Does NOT handle slice steppings.
    - Does NOT handle numpy.newaxis or None, either.
    - Does NOT support reducing the dimensionality of the data. (e.g. a[2:3] is okay, a[2] is not).
    
    (In principle, all of the above could be fixed...)
    """
    
    def __init__( self, data, slicing=(slice(None),) ):
        self._data = data
        slicing = self._expand_slicing(slicing, data.shape)
        slicing = self._explicit_slicing(slicing, data.shape)
        self._slicing = slicing

    def __array__(self):
        return self._data[self._slicing]
    
    @property
    def shape(self):
        return tuple(s.stop - s.start for s in self._slicing)
    
    @property
    def dtype(self):
        return self._data.dtype
    
    def __getitem__(self, slicing):
        assert all( isinstance(s, slice) for s in slicing ), "Sorry, SimpleView doesn't support integer slicing args yet."
        slicing = self._expand_slicing(slicing, self.shape)
        slicing = self._explicit_slicing(slicing, self.shape)
        combined_slicing = self._combine_slicings(self._slicing, slicing)
        return SimpleView(self._data, combined_slicing)
    
    def __setitem__(self, slicing, new_data):
        slicing = self._expand_slicing(slicing, self.shape)
        slicing = self._explicit_slicing(slicing, self.shape)
        combined_slicing = self._combine_slicings(self._slicing, slicing)
        self._data[combined_slicing] = new_data
    
    @classmethod
    def _combine_slicings(cls, slicing1, slicing2):
        assert len(slicing1) == len(slicing2)
        final_slicing = ()
        for s1, s2 in zip(slicing1, slicing2):
            # This code needs to be a little more complicated if we want to support steps other than 1.
            # For now, we just disallow that case.
            assert s1.step == None or s1.step == 1, "SimpleView does not support slice step sizes other than 1"
            assert s2.step == None or s2.step == 1, "SimpleView does not support slice step sizes other than 1"
            start = s1.start + s2.start
            stop = s1.start + s2.stop
            final_slicing += (slice(start, stop),)
        return final_slicing
    
    @classmethod
    def _explicit_slicing(cls, slicing, shape):
        """
        Replace all slice(None) items in the given slicing with 
          explicit start/stop coordinates using the given shape.
        Also, replace negative indexes with positive equivalents.
        """
        explicit_slicing = ()
[... 70 lines cut ...]
if __name__ == "__main__":
    import numpy
    a = numpy.random.random((100,100,100)).astype(numpy.float32)
    
    sv1 = SimpleView(a)
    (sv1[10:90, 20:80, 30:70][20:40, 40:60, :-1] == a[10:90, 20:80, 30:70][20:40, 40:60, :-1]).all()

    import h5py
    f = h5py.File('foo.h5', driver='core', backing_store=False, mode='w')
    f['dset'] = a
        
    sv2 = SimpleView(f['dset'])
    (sv2[10:90, 20:80, 30:70][20:40, 40:60, :-1] == a[10:90, 20:80, 30:70][20:40, 40:60, :-1]).all()

    # Try writing
    sv2[10:90, 20:80, 30:70][20:40, 40:60, :-1] = 1.0
    assert (f['dset'][:] == sv2).all()
    
    a[10:90, 20:80, 30:70][20:40, 40:60, :-1] = 1.0
    assert (f['dset'][:] == a).all()

    #print numpy.sum(sv1)
```

although the above handles the more general case that includes the entire slicing: syntax, colons, elipses, etc. What we need might be much simpler. Combining it with transpose might be tricky though.

mhhennig, 2019-08-30:
I just did a comparison, using a hdf5 file with some data:
```
f['recordings'].shape
(100, 9600000)
```
Here two ways to access the data. First read then transpose:
```
%%timeit
data = f['recordings'][10:20, 10000:100000].T
3.5 ms ± 946 µs per loop (mean ± std. dev. of 7 runs, 100 loops each)
```
and now transpose and then read:
```
%%timeit
data = f['recordings'][()].T[10000:100000, 10:20]
6.79 s ± 306 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```

Problem is that transposing does not create a view, but actually reads the whole data. In the biocam extractor, for instance, I therefore implemented views throughout (they also have different formats with every possible arrangement).

bendichter, 2019-08-30:
Oh really? I'll check that out!

bendichter, 2019-09-02:
@mhhennig Perhaps I am confused, but I thought that slicing an h5py array automatically reads the data, not creating a view, but a numpy array with the data loaded into ram.

For instance in this line:
https://github.com/SpikeInterface/spikeextractors/blob/e770e981b2d5389a23a70d71582f7809c20c9ae7/spikeextractors/extractors/biocamrecordingextractor/biocamrecordingextractor.py#L166

`rf['3BData/Raw'][nch * t1:nch * t0]` is not a view, but a numpy array, and the subsequent `.reshape` operates directly on the `np.array` object in a non-lazy way.

mhhennig, 2019-09-11:
@bendichter Yes, sorry, it reads the data. But transposing first reads *all* data, not just the segment you want. That's the difference here I think.

bendichter, 2019-09-17:
@mhhennig fyi: https://github.com/ben-dichter-consulting/lazy_ops

alejoe91, 2019-09-17:
Looks great @bendichter! Are you planning to release it soon? then we can definitely incorporate it! Otherwise we can just copy the file here. Let us know what you prefer

bendichter, 2019-09-17:
@alejoe91 We are planning to release it soon

bendichter, 2019-09-24:
@alejoe91 https://pypi.org/project/lazy-ops/

alejoe91, 2019-09-25:
@bendichter super! 

So we just have to substitute the hdf5 ops (NWB, Biocam, MEArec) with the lazy ops and that should work, right?

bendichter, 2019-09-25:
@alejoe91 Yes I would like to use it in those places, but it's not exactly a copy/paste situation. Unlike memory maps, the data in `h5py.Dataset`s (and by extension `DatasetViews`) is only accessible when the dataset is open. In places where we read data within a context, the end of that context closes the file, which means the data within the file on disk is no longer accessible. This is where implementing our own context managers (#232) come in. If we can create syntax like:

```python
with NwbRecordingExtractor('fpath') as re:
    data = re.get_traces()
    ...
```
we can make data accessible within that context and we can close the hdf5 file at the end of the context. 

If the RecordingExtractor is called not as a context I think we are better off reading the data the way we currently do, in a non-lazy way, first slicing then transposing.