SpikeInterface/spikeextractors #234: file name should not have the same names as the class include in it.

samuelgarcia, 2019-08-26:
In spike extractros we have
```from .RecordingExtractor import RecordingExtractor```
we should have file name in lower case to avoid name masking and so avoid this.

This is speculative but I think that this bug https://github.com/SpikeInterface/spikeinterface/pull/3 is due to this naming.
What do you think ?

alejoe91, 2019-08-26:
HI @samuelgarcia 

Maybe that's indeed the issue. We could have everything lower case to fix it. We would only need to change the RecordingExtractor, Multi.., Sub, Sorting.. and the __init__

colehurwitz, 2019-08-26:
We would have to change some stuff in spiketoolkit as well.

samuelgarcia, 2019-08-26:
@colehurwitz : In my mind spiketoolkit don't have file name with camelcase.

@alejoe91 : yes the fix is very fast.

Note that I am not sure at all that this is the problem. It is a guess.

alejoe91, 2019-08-26:
@colehurwitz in st we use `from spikeextracors import RecordingExtractor`, which will stay the same.

@samuelgarcia I recommend giving it a quick try :)

samuelgarcia, 2019-08-26:
I will. This my second item on todo list.
I will first  try the phinx gallery plugin :)

colehurwitz, 2019-08-26:
I mean, we need to change some file names in spiketoolkit as well. For example, the CurationSortingExtractor.

alejoe91, 2019-09-02:
spikeextractors: https://github.com/SpikeInterface/spikeextractors/pull/236

spiketoolkit: https://github.com/SpikeInterface/spiketoolkit/pull/227

closed 2019-09-02 by alejoe91