SpikeInterface/spikeextractors #235: request: MCS extractor

p2jones, 2019-08-26:
I'd like to use SpikeInterface with data recorded by amplifiers from Multi Channel Systems. Their data is stored in *.msrd or *.mcd files (or others, described in the [Multi Channel DataManager Manual](https://www.multichannelsystems.com/sites/multichannelsystems.com/files/documents/manuals/Multi%20Channel%20DataManager_Manual.pdf) but can be converted to H5 files by their [DataManager or command line tool](https://www.multichannelsystems.com/software/multi-channel-datamanager).

Is this simple to implement? I'd be happy to provide additional info or test files, or potentially tackle this myself with some guidance. Is there a guide on creating a spikeextractor?

colehurwitz, 2019-08-26:
Hi @p2jones ,

Implementing an extractor is actually pretty straightforward. 

All you have to do to create a new RecordingExtractor is to implement the abstract methods in:
https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/RecordingExtractor.py

To create a new SortingExtractor,  you have to implement the abstract methods in:
https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/SortingExtractor.py

In both these cases, if you can implement a write function that allows you to write other extractors into your underlying format, that would be great. It is not required though. Also, if you want to add a unit test, that would also be helpful.

To see examples of current extractor implementations, you can check:
https://github.com/SpikeInterface/spikeextractors/tree/master/spikeextractors/extractors

If you want to leave a pull request with your proposed extractor, we can check to make sure everything is done in accordance with other extractors.

If you have any questions, please feel free to leave an issue or comment on this one. :)

colehurwitz, 2019-08-26:
@alejoe91 Indeed!

alejoe91, 2019-08-26:
Hi @p2jones 

It is quite straightforward to implement an extractor, and if you are familiar with the data format I think it will be very easy for you to do it. 

There is a "Building a new RecordingExtractor/SortingExtractor" section in the README. In brief, for a `RecordingExtractor`, you can inherit from the base [`RecordingExtractor `](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/RecordingExtractor.py) class and you have to implement the following methods:

- `get_sampling_frequency()`: returns sampling frequency in Hz (float)
- `get_traces()`: returns signals as an M channels x T timestamps array (or h5 object)
- `get_channel_ids()`: returns a list of channel ids

There are a large variety of extractors available. You can browse them [here](https://github.com/SpikeInterface/spikeextractors/tree/master/spikeextractors/extractors).

Let us know if you need further assistance :)

alejoe91, 2019-08-26:
@colehurwitz good timing!

p2jones, 2019-08-26:
Ok I'll have a shot at it. Thanks!

colehurwitz, 2019-08-26:
_gui_params is not required, but would be nice to have and very easy to make. It is a list of dictionaries for each param in your subextractor's initialization (the format can been seen in the example I set). It allows users to use extractors from within a GUI.

If you don't want to make it or are confused, I can put it in afterwards.

p2jones, 2019-08-26:
Is `_gui_params` necessary?

I've managed to make a new MCSRecodingExtractor in a local branch (MCS). I tried to push (`git push -u origin MCS`) and got:

`remote: Permission to SpikeInterface/spikeextractors.git denied to p2jones`

What's the best way to do this? I haven't contributed to something like this before.

alejoe91, 2019-08-26:
You should fork the repo and push the changes on your fork. Then you can make a pull request from your fork to this repo!

p2jones, 2019-08-26:
Forked and pushed: https://github.com/NMI-MSNT/spikeextractors/tree/MCS

I based it on the Biocam extractor, so perhaps I have unnecessary structure in the extractor. For example, I don't know if openMCSFile should be a separate function, or could just be in __init__.

Should I do a pull request in this state?

https://github.com/NMI-MSNT/spikeextractors/blob/MCS/spikeextractors/extractors/mcsrecordingextractor/mcsrecordingextractor.py

It would be useful to load some kind of layout file to know the physical location of each electrode. This doesn't seem to be in the H5 files. I would talk to MCS to see the best way of adding this functionality.

alejoe91, 2019-08-26:
You can make the pull request and we comment there if we have any doubts :) for now the probe layout can be easily loaded as a prb or a CSV file using the `load_probe_file` function of spikeextractors

samuelgarcia, 2019-08-26:
Hi @p2jones.
I have implemented recently the MSC in neo.
See https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/rawmcsrawio.py

I will soon start a generic wrapper on top of neo in spikeextractor. So all format available in neo.rawio willl be available inside spikeinterface.
The advantage is that you won't need conversion to hdf5 with there tools, you could acces directly with lazy load the signals from MSC format.

The actual MSC is in neo is not extensivly tested because quite recent.
Maybe you could give it a try.

samuelgarcia, 2019-08-26:
@p2jones : could you named you extractor MCSHDFRecordingExtractor instead of MCSRecordingExtractor because MCSRecordingExtractor name could fit better the natural format when it will be implemented.

samuelgarcia, 2019-08-26:
My bad, the link I send you is the OLD MSC format from the old version.

the new one, is implemented in a branch here https://github.com/samuelgarcia/python-neo/blob/format_msrd/neo/rawio/msrdrawio.py

This new link implement the NEW msrd format from multi channel that replace the old MSC format. The previous I send you is the the MSC converted to custulmized bin format and not maintain anymore by multi channel.

Best

p2jones, 2019-08-27:
Ok, I've renamed it to MCSH5RecordingExtractor, and fixed a couple bugs.

Thanks @samuelgarcia for the info about neo. I'll have to try that out as well.

MCS data can include arbitray unordered channel IDs, for example, if a few channels are exported from a larger recording. In MCSH5RecordingExtractor, I've now included `channel_ids` (0 to n-1) as well as `channel_ids_MCS` which can be arbitrary. `channel_ids` seems to be important for some built in functions (e.g. `sw.plot_timeseries`). Is there a better way of solving this?

alejoe91, 2019-08-27:
@p2jones in the plot_timeseries it is only used to select some channels for plotting. It is perfectly fine to return (0 to n-1)

p2jones, 2019-08-28:
Ok. I misunderstood a bug that I was having. It's fine now with a arbitrary channel_ids (for example, a recording with six channels with IDs of 15, 17, 26, 44, 45, and 58). It seems to be working now.

alejoe91, 2019-08-28:
Great! Can you make a pull request so we can compare the code and merge when it's ready? After you make a PR, you can still commit changes and we can review the code :)

colehurwitz, 2019-08-30:
One question @p2jones . Do you need mea_pitch? It isn't used anywhere in the extractor.

p2jones, 2019-08-30:
No, it can be removed.

colehurwitz, 2019-08-30:
Thanks so much for implementing this extractor! We appreciate the contribution and hopefully you find it useful for you work. :)

closed 2019-08-30 by colehurwitz