SpikeInterface/spikeextractors #241: soft clustering

bendichter, 2019-09-02:
Is it possible to represent soft clustering of spikes in the SpikeExtractor API? Some sorters output a probability that a spike belongs to a few different clusters. You could just take the max and move on, but some labs are starting to prefer to treat these spikes probabilistically, accounting for the fact that it is uncertain which cluster it truly belongs to. Would it be possible to support this types of data?

https://github.com/NeurodataWithoutBorders/pynwb/issues/1056

colehurwitz, 2019-09-02:
I think this can be handled by spike features in the SortingExtractor. Spike features are stored in dictionaries (like unit properties) but are arrays of values that correspond to the spike times in each SortingExtractor. Each spike could have an associated probability feature.

bendichter, 2019-09-02:
@colehurwitz sounds reasonable. This would require a decent amount of data duplication, but may be the best approach nonetheless.

alejoe91, 2019-09-02:
Hi @bendichter 

As @colehurwitz it is possible [spike features](https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/sortingextractor.py#L88). 

Features are a list or an array with the same number of elements as the spike train. So I would assign the unit to the highest probability cluster, and then you could have a `probability` feature which can be a list if dictionaries. For example:

```
sorting.get_unit_spike_train(1)

[100, 200]

probabilities = [[dict('p_u1': 0.7), dict('p_u2': 0.1), dict('p_u3': 0.4)],
                         [dict('p_u1': 0.8), dict('p_u2': 0.2), dict('p_u3': 0.5)]]

sorting.set_unit_spike_feature(unit_id=1, feature_name='probability', value=probabilities)
```

Not he most straightforward way..but it could work :)

colehurwitz, 2019-09-02:
@alejoe91 If you check his example, he mentioned that each unit would have duplicate spikes and then probabilities can be stored 1-1 without dictionaries.

bendichter, 2019-09-02:
@colehurwitz Yes that's true but I'm just trying to think through different options for storing this information.

@alejoe91 Yes I like your proposed system. I like that if a user is naive to this whole probability system, then the system defaults to just using the max probability cluster, which seems like the right default mode. Are the strings 'p_u1' already used as unit identifiers? If not, it seems like units ints as the keys may be more straightforward.

colehurwitz, 2019-09-02:
I like Alessio's method as well of storing the spike time in the max probability unit spike train. I think having units ints as keys would be fine.