SpikeInterface/spikeextractors #244: get_shared_unit_property_names() method for SortingExtractor

luiztauffer, 2019-09-06:
Suggestion:
The recent update on RecordingExtractor giving the method get_shared_channel_property_names() was very nice, how about doing a similar method for SortingExtractor, e.g. get_shared_unit_property_names() ?

alejoe91, 2019-09-06:
I totally agree :) We should definitely do that. It might take a few dates as we have to update all packages.

colehurwitz, 2019-09-06:
Good catch. I’ll do this today.

colehurwitz, 2019-09-06:
#245 

I also updated spiketoolkit and spikewidgets to match this new functionality (all other repos did not use these functions).

samuelgarcia, 2019-09-06:
You shoudl update spikeinterface/examples.
examples elsewhere should be removed.

colehurwitz, 2019-09-06:
I updated the spikeinterface examples

colehurwitz, 2019-09-06:
Should I remove all examples from the other repos? Just delete the examples folders?

samuelgarcia, 2019-09-06:
@alejoe91 ?

colehurwitz, 2019-09-06:
Alessio just told me to delete the folders. Will do

closed 2019-09-06 by colehurwitz