SpikeInterface/spikeextractors #247: Make load probe file a function of RecordingExtractor

colehurwitz, 2019-09-07:
We can make 'load_probe_file" as a function of the RecordingExtractor. More object-oriented and more natural in my opinion.

`recording.load_probe_file('tetrode.prb')`

colehurwitz, 2019-09-07:
@alejoe91

alejoe91, 2019-09-07:
the load_probe_file can reorder channels so it returns a subrecording extractor. Also the probe file might exclude some channels so I don't think this can be handled by the class.

colehurwitz, 2019-09-07:
Hmmm. So I think the name is pretty misleading right now. I always thought that it was loading properties into the original extractor (given the name and how we use it). However, it never modifies the original extractor, it only returns a new extractor.

So:

```
recording = MyRecordingExtractor('path')
recording_prb = se.load_probe_file('tetrode.prb')

recording != recording_prb
```

colehurwitz, 2019-09-07:
I think the name should be changed to represent that we are giving a new extractor with the loaded properties: `get_loaded_probe_extractor` or something.

colehurwitz, 2019-09-07:
Or we actually load the probe information by changing the recording extractor's channel_ids and reordering them.

alejoe91, 2019-09-07:
For removing and reordering channels we need a subrecording extractor. Maybe we could change the name but as long as we document it properly I don't see this as a big problem. Returning another sub extractor makes it more flexible IMO

colehurwitz, 2019-09-07:
Ok, I agree with you. Let's just change the name then. How does this look:

```
recording = MyRecordingExtractor('path')
recording_prb = se.create_probe_subrecording(recording, 'tetrode.prb')

recording != recording_prb
```

colehurwitz, 2019-09-07:
Btw, I think this will be much easier if/when we move to a Probe object that is stored inside a RecordingExtractor (someday). Then we could have an actual load probe function that modifies that internal object which can reorder and get rid of channels in the extractor.

alejoe91, 2019-09-07:
Mmm let's discuss it with the others. Personally I like the load_probe_file name and it's clear that another recording extractor is returned. 
recording will be different than recording_prb anyways as properties are loaded. I don't think it makes a difference for the user

alejoe91, 2019-09-07:
I agree with the probe object. It's in the plans :)

colehurwitz, 2019-09-07:
I think the name is just a bit too confusing.

A person that is quite familiar with object-oriented programming would probably want to do this:

```
recording = MyRecordingExtractor('path')
se.load_probe_file(recording, 'tetrode.prb')
```

alejoe91, 2019-09-07:
It's written in the docs. It returns another extractor :) I think people will read the docs or the docstrings and it'll be clear

alejoe91, 2019-09-07:
It's not a member function of the object in fact, but an external function

colehurwitz, 2019-09-07:
But returning another extractor does not imply loading anything. I think it is very confusing.

I actually use this function a lot and I didn't know that it did this. I think we should change it to `create_probe_subrecording` until we actually load stuff. We can discuss with @samuelgarcia @magland .

alejoe91, 2019-09-07:
Mmm I don't like the name honestly. Let's discuss with the others. I think that if we document it properly (what I'd does is written in the docstrings and the docs) it should be fine for now (until we have a probe abstraction).

alejoe91, 2019-09-07:
Alternatively we can set do: recording = subrecording, without returning anything.

colehurwitz, 2019-09-07:
That last idea is interesting... I would like it if that works. But I am not sure you can have a recording = subrecording because the subrecording references the recording! haha

alejoe91, 2019-09-07:
We can try..I agree it would remove some confusion. Let's see tomorrow

samuelgarcia, 2019-09-09:
I think the debate here is "inplace modify" or "new object".

As a pandas user, that offer the 2 possibilities. At the begening I was happy with the inplace modified object but in fact it is a trap.
I now like very more the semantic ```new_object = object.do_something()```

In short, I am happy with the actual behavior.

About the name, I anderstand the "load" term is alittle bit confusing because we coud imagine that it is inplace.
Maybe it could:
```recording_prb = se.modify_with_probe_file(recording, 'tetrode.prb')```

colehurwitz, 2019-09-09:
I am ok with the functionality as well. Do you think it might be better as an object function though (extractors create new subextractors with probe files)?

Also, I’m not sure I like the name you recommended as well because it still implies the original object is being changed (load, modify, etc). I think we need a name that is clear that nothing is happening to the original object.

colehurwitz, 2019-09-14:
#248

closed 2019-09-14 by colehurwitz