SpikeInterface/spikeextractors #249: Suggestion: Tutorials directory

luiztauffer, 2019-09-10:
Hi, together with the NWB extractors extension I prepared a Jupyter notebook tutorial on how to use SpikeExtractors with NWB files.
I was wondering if you guys would like to have it somewhere, e.g. a tutorials directory.

colehurwitz, 2019-09-10:
We have a [spiketutorials](https://github.com/SpikeInterface/spiketutorials) package that might be a good fit for this! We have one workshop tutorial for building general pipelines, but it would be nice to have a good tutorial about integration into NWB.

samuelgarcia, 2019-09-10:
We also have tutorial (generated to notebook) here in spikeinterface meta package:

[source](https://github.com/SpikeInterface/spikeinterface/tree/master/examples)
and
[tutorial galery](https://spikeinterface.readthedocs.io/en/latest/modules/index.html)

alejoe91, 2019-09-10:
I think it would be nice it to have it in the docs. @luiztauffer is the notebook on github?

luiztauffer, 2019-09-11:
not yet, but I can PR it to the spiketutorials repo and you guys can decide if it's worth it to put on documentation or anywhere else 
:)

alejoe91, 2019-09-11:
Sounds good! :)

alejoe91, 2020-06-19:
@luiztauffer an update on this.

We have now a github.io page (https://github.com/SpikeInterface/SpikeInterface.github.io) that uses nikola and it allows to have several notebooksin the same place: https://spikeinterface.github.io/

Probably this would be a better place to have an SI-NWB tutorial

closed 2021-03-11 by alejoe91