SpikeInterface/spikeextractors #251: Exporter Object

colehurwitz, 2019-09-10:
I have been braining storming with Roger about an `Exporter` object which I feel might be important to have. Currently, for convenience, we keep all export code as a static function in an `Extractor`. I was thinking that we could keep this code there, but also build an object around the static function that is just for exporting. Then the user could import and use an `Exporter` object which is more intuitive (and easier to put in applications like the GUI).

The challenge is that one of the export functions is in spiketoolkit (`export_to_phy`)... so I am not sure where the `Exporter` baseclass would live and function. I feel like it should be in spikeextractors, but then it would depend on spiketoolkit (creating a cyclical dependency). Also, we have two export_to_phy functions currently (from the PhySortingExtractor and the export_to_phy) function, which repeats code and can be a bit confusing (although I know the difference, a user may not).

My proposal for an Extractor object would be something like:

```
RecordingExporter:
    def __init__(self, recording, save_path):

SortingExporter:
    def __init__(self, sorting, save_path):
```
Then the `export_to_phy` class might inherit from both of these options:
```
PhyExporter(SortingExporter, RecordingExporter)
```

What are your thoughts, @alejoe91 @magland @samuelgarcia ?

alejoe91, 2019-09-10:
What would be a sample use case that can't be done by the write_recording function (exactly same parameters as the unit you suggest)? Would you need an exporter for each class then? I think that you should be able to export without the need to instantiates another object only for that purpose, no? In my mind creating an extra object just makes it more convoluted. 

Regarding phy, the export function needs post processing, so it's a bit different.. I don't think we should create a cyclic dep just for this.

If an Exporter is handy for spikely, why don't you make it there? :)

colehurwitz, 2019-09-10:
It could allow extra exporting functionality. For example, we could implement something like an `export_by_property` function that creates subrecording extractors and exports each one separately. This is just an example, but generally, having an object to encapsulate a processing step is what we do in SpikeInterface and it allows for added functionality down the line.

colehurwitz, 2019-09-11:
Hmm, I could be persuaded against this. Not an important issue to be honest

colehurwitz, 2019-09-14:
This is probably not necessary.

closed 2019-09-14 by colehurwitz