SpikeInterface/spikeextractors #252: Directories and Files in the IO system

colehurwitz, 2019-09-11:
In our extractors, the name of the file or the directory is different depending on the subclass and sometimes it is confusing if it needs a directory or a file.

I was thinking we could do something similar to NEO (page 19): https://buildmedia.readthedocs.org/media/pdf/neo/0.4.1/neo.pdf

That will make integration with NEO easier in the future as well @alejoe91 @samuelgarcia

samuelgarcia, 2019-09-11:
:)
Everything should be like neo :)
Note that 0.4.1 is very very old version.
We are now in 0.7.2
https://neo.readthedocs.io/en/0.7.2/
But I think for that stuff it haven't change.

colehurwitz, 2019-09-11:
Also:

is_readable
is_writable

is very nice.

colehurwitz, 2019-09-11:
Although we can't switch to NEO now, maybe we can prepare our API now a bit.

colehurwitz, 2019-09-11:
How does this look:

```
class RecordingExtractor(ABC):
    extractor_name = ''
    has_default_locations = False
    is_readable = False #instead of installed = False
    is_writable = False
    _gui_params = []
    installation_mesg = ""  # error message when not installed
    mode = 'file'                 #'dir' or 'file_or_dir"
```

For the init function, how can we ensure that each mode has the correct keyword, like this:
```
def __init__(self, **kargs):
```

samuelgarcia, 2019-09-11:
Yes we can prepare.
The neo integration will be very slow....
I need first that neo 0.8 is released which is not the case....

colehurwitz, 2019-09-11:
How did you make sure that file and directory were required arguments given the mode?

samuelgarcia, 2019-09-11:
I think in each case, the reader have karg either dirname='XXX' or filename='XXX'.
This mode is a hint for GUI stuff.

colehurwitz, 2019-09-11:
Got it. I will make some changes to the code today.. Good to hammer down some API stuff before the paper releases.

colehurwitz, 2019-09-14:
#254

closed 2019-09-14 by colehurwitz