SpikeInterface/spikeextractors #263: MCSH5 Extractor newer format doesnt work

McD-NMI, 2019-09-22:
@colehurwitz 
It seems that the MultiChannel System (MCSH5) extractor can only handle H5 files from older versions of MCS software, and does not appear to work with their new H5 format. 

Any plans to update for the newest format as well? 

When I run older version it works fine, but I get this error in newer format. 

> Traceback (most recent call last):
>   File "C:/Users/mattm/PycharmProjects/OrganoidMEA/Spyke_sorting/OrgMEA_spikes.py", line 14, in <module>
>     recording = se.MCSH5RecordingExtractor(recording_file)
>   File "C:\Users\mattm\PycharmProjects\OrganoidMEA\venv\lib\site-packages\spikeextractors\extractors\mcsh5recordingextractor\mcsh5recordingextractor.py", line 28, in __init__
>     self._recording_file, verbose)
>   File "C:\Users\mattm\PycharmProjects\OrganoidMEA\venv\lib\site-packages\spikeextractors\extractors\mcsh5recordingextractor\mcsh5recordingextractor.py", line 95, in openMCSH5File
>     assert Unit==b'V', 'Unexpected units found, expected volts, found {}'.format(Unit.decode('UTF-8'))
> AssertionError: Unexpected units found, expected volts, found NoUnit
> Exception ignored in: <function MCSH5RecordingExtractor.__del__ at 0x0000017AF3E6A438>
> Traceback (most recent call last):
>   File "C:\Users\mattm\PycharmProjects\OrganoidMEA\venv\lib\site-packages\spikeextractors\extractors\mcsh5recordingextractor\mcsh5recordingextractor.py", line 32, in __del__
>     self._rf.close()
> AttributeError: 'MCSH5RecordingExtractor' object has no attribute '_rf'

colehurwitz, 2019-09-23:
We would definitely like to support all versions of the MCS H5 format both old and new! 

We are currently preparing for a big release (along with a preprint) so things are pretty hectic right now. It would be great if you could help us update our mcsh5recordingextractor to be compatible with the newer format. Could you help us with that and then make a pull request we could review?

McD-NMI, 2019-09-24:
Hmmm, I think I was initially wrong about this and there shouldnt be a difference in the .h5 files generated by .mcd and .msrd file types. Just a mistkae on my part. However I did end up looking through the extractor and will comit some small changes soon anyway!

closed 2019-09-24 by McD-NMI

colehurwitz, 2019-09-24:
Great! Thanks for looking through our implementation and for your upcoming changes. :+1: