SpikeInterface/spikeextractors #281: NwbExtractor fails as session and identifier are not set

alejoe91, 2019-10-24:
In `write_recording` and `write_sorting`, `session` and `identifier` are set to `''`. However, this results in an Error when attempting to open the file with pynwb or when writing sorting data on a file created by `write_recording`.

Suggested fix: 
- add `session` and `identifier` as optional args in the write functions
- if None (by default) use `session_0` and a random identifier.

@bendichter @luiztauffer does this sound reasonable?

alejoe91, 2019-10-24:
Also this line: https://github.com/SpikeInterface/spikeextractors/blob/master/spikeextractors/extractors/nwbextractors/nwbextractors.py#L183

Should be:
spkt = sorting.get_unit_spike_train(unit_id=id) / fs

Making a PR now

alejoe91, 2019-10-24:
https://github.com/SpikeInterface/spikeextractors/pull/282

seancav1967, 2019-10-25:
Please remove all bugs

bendichter, 2019-10-26:
@alejoe91 yes, sounds great

closed 2019-10-26 by alejoe91