SpikeInterface/spikeextractors #290: Provenance information records "interface"

yarikoptic, 2019-11-01:
It would be great if provenance information about algorithm and its parameters, or even possibly loose human description (from manual curation) was recorded alongside with the actual outputs of spike extractors.  Are there already such records available in any file format/extractors "output/protocol" and could be transferred/converted from one format to another?

In NWB there is only `/general/source_script` which could potentially, and at least temporarily, encapsulate such information if the only thing in the file is spike sorted data.  But a more generalized approach (e.g. based on PROV standard) is "in discussion" ATM: https://github.com/NeurodataWithoutBorders/nwb-schema/issues/319
And there I bring an example of  [ASDF, the Adaptable Seismic Data Format](https://asdf-definition.readthedocs.io) which is HDF5-based and stores [SEIS-PROV](http://seismicdata.github.io/SEIS-PROV) [provenance records under /Provenance](https://asdf-definition.readthedocs.io/en/latest/provenance.html#provenance).

I think it would be valuable if SpikeInterfaces/* started to envision channeling such provenance information and may be even spike sorted data could be something for which we could initiate something like NWB-PROV; and meanwhile store some detectable serialization within `/general/source_script`

colehurwitz, 2019-11-01:
This is a good idea. I imagine that the spikesorters can store the provenance in the SortingExtractors upon completion. Then SortingExtractors can save the provenance.

colehurwitz, 2019-11-04:
I was thinking something like a `sorting_provenance` dictionary in a SortingExtractor that is empty by default, but can be added to by sorting algorithms. 

What NWB decides to do with this information (PROV, etc.) can be done outside of spikeextractors with this dictionary then?

colehurwitz, 2019-11-04:
I am afraid that I am not too familiar with ASDF or PROV so I couldn't follow the thread you referenced very well.

yarikoptic, 2019-11-04:
> I am afraid that I am not too familiar with ASDF or PROV so I couldn't follow the thread you referenced very well.

neither am I . Learning as I go ;)

I am afraid that it might take awhile to get provenance formalized at official NWB level. Meanwhile it would be nice and sufficient to store the record as some ad-hoc record so later we could have recovered all needed information and converted to proper nwb provenance.
ATM a single legit place with .nwb seems to be the `/general/source_script` field:

```
core/nwb.file.yaml-    - name: source_script
core/nwb.file.yaml-      dtype: text
core/nwb.file.yaml-      doc: Script file used to create this NWB file.
core/nwb.file.yaml-      attributes:
core/nwb.file.yaml-      - name: file_name
core/nwb.file.yaml-        dtype: text
core/nwb.file.yaml:        doc: Name of script file
core/nwb.file.yaml-      quantity: '?'
```
so I think it would be legit to (ab)use it for the purpose here especially if sorting would go into a separate .nwb file. It could even be made nearly 100% legit while allowing to have sorting spec coded in so we could get to it later programatically. E.g. if that `sorting_provenance` dict (I will call it a `sorting_spec` below) could be consumed by some high level function in spikeextractors one way (as below) or another (via **kwargs), then it could be:

```python
#!/usr/bin/env python

sorting_spec = {
 ... parameters go here ....
}

if __name__ ==  '__main__':
    infile, outfile = sys.argv[1:3]
    from spikeextractors import run_spike_sorting
    run_spike_sorting(infile, sorting_spec, outfile)
```
content within `/general/source_script` which we could either parse or just import later on to get `sorting_spec` out. Just an idea ;) or too ad-hoc?

colehurwitz, 2019-11-04:
I think `sorting_spec` and `software_prov' can be created automatically by a spike sorter and the stored in the sorting extractor and also written into the output folder:

```
import spikesorters as ss
import spikeextractors as se

recording = se.MyFormatRecordingExtractor('path-to-recording')
sorting = ss.run_mountainsort4(recording, output_folder='path-to-output')

#Access the two provenance dictionaries
sorting.software_prov
sorting.sorting_spec
```

colehurwitz, 2019-11-04:
Btw, spikeextractors just has file IO. The spikesorters repository has all spike sorting algorithms and code.

yarikoptic, 2019-12-02:
> Btw, spikeextractors just has file IO. The spikesorters repository has all spike sorting algorithms and code.

which makes it even more fun as to how much provenance information it is needed to record ;)

samuelgarcia, 2019-12-03:
Hi all.
One key for this is the serialization into json (or equivalent) of
extractors.

I already need this feature for the launcher to create a queue task list.

alejoe91, 2020-03-30:
@yarikoptic 

We made some progress in provenance! now each extractor has an attached dictionary that keeps track of everything that the extractor goes through (by nesting several dictionaries with kwargs of each processing step). This makes it possible to dump to json any extractor and load it back quickly. Similarly, each spike sorter now saves a json file with its version and params.

yarikoptic, 2020-03-30:
That is great @alejoe91 , thank you! Could you point me to some quick example or just paste it here to appreciate the new feature? ;)

alejoe91, 2020-03-30:
I'll update the docs in a few days :)

In brief, at any point in the pipeline you can call (same holds for SortingExtractors):
```
d = recording.make_serialized_dict()
recording.dump('some-json-path.json')
```
The dictionary will contain the `class` name, `module` (spikeextractor or spiketoolkit), `version`, and `kwargs`. Note that if a keyword arg is an extractor, we dump the serialized dict of that extractor.

You can reload the extractors with:
```
extractor = se.load_extractor_from_dict(d)
extractor = se.load_extractor_from_json('some-json-path.json')
```

This mechanism also allowed us to speed up processing using joblib (making a dictionary is like pickling more or less)!

Finally, spike sorting runs now saw a `spikeinterface_log.json` (with sorter name, version, run time, and return value - 0 for success) and a `spikeinterface_params.json`, with all the params used as input to the sorter.

yarikoptic, 2020-03-30:
Cool.  Pardon my ignorance -- does it mean that every `sorting` output (like in an example above) would have enough of provenance information (also in a data file when stored) to produce such provenance record? (I could just wait for documentation piece to come so you don't need to preclude it by answering here)

alejoe91, 2020-03-30:
Yes! As long as the underlying file/folder is not deleted. Currently this is supported for all extractors, preprocessors, and automatic curators (e.g. removing units below some SNR). Basically the provenance record is the json file. Note that the transition between the recording and the sorting can be reproduced using the log and params files generated by the spike sorting run :)

closed 2020-04-22 by alejoe91