SpikeInterface/spikeextractors #304: Spyking Extractor data read

Marrekko, 2019-11-18:
Hey there,

first thanks for the great work on this project! I'm new to python based analysis, so excuse me as that might be a minor issue.  I want to use the spikeinterface toolkit for postprocessing and other things. However, I'm having problems to read my Spyking Circus output using the Spyking Extractor.

> import spikeinterface.extractors as se
> import spikeinterface.widgets as sw
> recording,sorting= se.SpykingCircusRecordingExtractor("G:/Daten/35340/1")

throws me a

> TypeError                                 Traceback (most recent call last)
> <ipython-input-32-630a3b211a10> in <module>
> ----> 1 recording,sorting= se.SpykingCircusRecordingExtractor("C:/Users/Marek/Desktop/35340/")
> 
> ~\Anaconda3\lib\site-packages\spikeextractors\extractors\spykingcircusextractors\spykingcircusextractors.py in __init__(self, folder_path)
>      51             if f.suffix == '.npy':
>      52                 recording_file = str(f)
> ---> 53         NumpyRecordingExtractor.__init__(self, recording_file, sample_rate)
>      54 
>      55 
> 
> ~\Anaconda3\lib\site-packages\spikeextractors\extractors\numpyextractors\numpyextractors.py in __init__(self, timeseries, sampling_frequency, geom)
>      20             self._timeseries = timeseries
>      21         else:
> ---> 22             raise TypeError("'timeseries must be a .npy file name or a numpy array")
>      23         RecordingExtractor.__init__(self)
>      24         self._sampling_frequency = float(sampling_frequency)
> 
> TypeError: 'timeseries must be a .npy file name or a numpy array

So I wonder, why the SKC Extractor works with the Numpy Extractor in the end, as the native output are hdf5 and of course, no npy file of the timeseries is generated. I would be very thankful for a little hint how to make this work!

alejoe91, 2019-11-18:
Hi @Marrekko 

Spykin Circus supports several input formats. In our python wrapper we use npy format so the RecordingExtractor expects that. I agree that this could be improved to read the format from the params and use the right extractor accordingly. What type of input format do you use?

Note that the hdf5 format is used for the sorted data. The input data are provided externally. If you used the binary format you can load the recording `se.BinDatRecordingExtractor`, by providing a few arguments (sampling frequency, number of channels, dtype).
You can load the spike sorted data using the `se.SpykingCircusSortingExtractor`, that reads from the hdf5. Let me know if that works!

samuelgarcia, 2019-11-18:
Hi @Marrekko.
I think you are mixing up RecordingExtractor  and SortingExtractor.
hdf5 is the internal format for SC.

The input of spykingcircus wrapper is a always RecordingExtractor (all format).
In any case internal the recording is copied inside the working the folder. 
So the input format (npz) do not matter so much.

alejoe91, 2019-11-18:
@samuelgarcia I think @Marrekko was trying to import some of his already sorted data in spikeinterface. We also have a SCRecordingExtractor, but it assumes that the data have been sorted with spikeinterface

samuelgarcia, 2019-11-18:
OK sorry.

Marrekko, 2019-11-18:
Hey Alessio and samuel,
Thanks for your superfast Answer!
Yes, I'm working with MCS generated binary data and reading of the recording with `se.BinDatRecordingExtractor` works fine, as well as reading of the sorted data with the `se.SpykingCircusSortingExtractor` . Pretty simple to just split that up manually. Solved.
Thank you!
PS: Yeah would be cool to give the SCSExtractor the ability to readout the input file format and choose the Extractor accordingly, that was my major confusing point.

alejoe91, 2019-11-18:
@Marrekko great that it's solved. Your input data actually are not in SC format. They are binary and SC can read it. The SCRecordingExtractor was designed to retrieve the recording and sorted data from the folder created using spikeinterface, that's why it's so specific. I think it's a good practice to decouple the loading of the recording and the sorting. 

Ler us know if you find other issues! Closing this

closed 2019-11-18 by alejoe91