SpikeInterface/spikeextractors #310: request: Simple examples on Docs for loading custom data (i.e NumpyRecordingExtractor/NumpySortingExtractor)

sbtnRey, 2019-12-06:


mhhennig, 2019-12-15:
How do you want to use this, what would be useful to show?

sbtnRey, 2019-12-17:
Primarily how custom data should be formatted for the numpy extractor (i.e data structure format for channel values)

alejoe91, 2019-12-18:
Hi @sbtnRey 

If I understand correctly, you have data loaded in memory (a numpy array) and you would like to create a NumpyRecordngExtractor object? 

In this case, you can check out [this tutorial](https://spikeinterface.readthedocs.io/en/latest/modules/extractors/plot_1_recording_extractor.html#sphx-glr-modules-extractors-plot-1-recording-extractor-py). You need to provide the timeseries and sampling frequency, and optionally the geometry. 

Is this what you need?

sbtnRey, 2019-12-23:
Yeah, that was exactly what I needed. Thank you!

closed 2019-12-25 by alejoe91