SpikeInterface/spikeextractors #318: Openephys parallel

benefron, 2020-02-02:
Hi,
Is there a way to run the extractor in parallel for each electrode file?
Ben

alejoe91, 2020-02-02:
Hi Ben, 

Unfortunately no. I highly recommend saving in binary format instead if openephys so that the data are memmapped.

Cheers
Alessio

benefron, 2020-02-02:
so change data I already have to binary? so it can run efficiently on HPC?

alejoe91, 2020-02-02:
I don't think you can change the data you have to binary. But if you will acquire more data I recommend binary :)

benefron, 2020-02-02:
what about this in the extractors:
write_to_binary_dat_format(self, save_path, time_axis=0, dtype=None, chunk_size=None):
     '''Saves the traces of this recording extractor into binary .dat format.

alejoe91, 2020-02-03:
Hi @benefron unfortunately that won't work, because in order to write_to_binary you have to load the data into memory. Then you can save to binary and if you load it another time this will speed up things a lot

benefron, 2020-02-03:
It might be good for me. I will save it to binary on my desktop and on the HPC run the binary file.
Thanks

closed 2020-02-03 by benefron